pycom/pycom-libraries #11: LoRaWAN Nanogateway for AS923

sh4nnongoh, 2017-05-28:
Hi,

Is there a working example for interaction with the Things Network using AS923?

I have 2 LoPys with each acting as an OTAA node and a nano gateway node respectively.
I modified the following files as follows,

config.py:
```
SERVER = 'router.as.thethings.network'
LORA_FREQUENCY = 923200000 
LORA_DR = "SF10BW125" 
```

nanogateway.py:
```
RX_PK = {"rxpk": [{"time": "", "tmst": 0,
                   "chan": 2, "rfch": 0,
                   "freq": 923.2, "stat": 1,
                   "modu": "LORA", "datr": "SF10BW125",
                   "codr": "4/5", "rssi": 0,
                   "lsnr": 0, "size": 0,
                   "data": ""}]}
```

OTAA/main.py:
```
for channel in range(0, 72):
    lora.remove_channel(channel)

for i in range(0, 16):
    lora.add_channel(i, frequency=923200000, dr_min=0, dr_max=4) #not sure why cannot set DR5
```

The JOIN_REQUEST & JOIN_ACCEPT messages have been successfully generated. However, my OTAA node is unable to join the network.

[image]

Please advise, thank you.

Bucknalla, 2017-07-19:
Hi @sh4nnongoh could you please share your OTAA node, code?

Thanks!

sdaniel55, 2018-07-05:
Hi @sh4nnongoh, I'll close this issue due to inactivity. If you're still having problems, feel free to reopen it. Thanks!

closed 2018-07-05 by sdaniel55