pycom/pycom-libraries #13: Pysense accelerometer example (library LIS2HH12): I2C bus error

jellium, 2017-06-06:
Import [lis2hh12.py](https://github.com/pycom/pycom-libraries/blob/master/pysense/LIS2HH12/lis2hh12.py) into a LoPy drive in directory `/flash/lib/`

Create a file containing [the following code](https://docs.pycom.io/pycom_esp32/pycom_esp32/tutorial/includes/pysense-examples.html#axis-accelerometer):
```
from machine import I2C
from lis2hh12 import LIS2HH12
import time

i2c = I2C(0, baudrate=100000, pins=('P9', 'P10'))      # Initialize the I2C bus

acc = LIS2HH12(i2c=i2c)

while True:
    print('----------------------------------')
    print('X, Y, Z:', acc.read())
    print('Roll:', acc.roll())
    print('Pitch:', acc.pitch())
    print('Yaw:', acc.yaw())
    time.sleep(1)
```
Produces the following error:
```
Traceback (most recent call last):
    File "<stdin>", line 8, in <module>
    File "/flash/lib/lis2hh12.py", line 37, in __init__
OSError: I2C bus error
```

Thanks for your work & help!

Bucknalla, 2017-06-06:
Hi @jellium, this is a placeholder library that was designed for the pre-production version of the Pysense/Pytrack. The official library will be releasing at the end of this week and will correct the Pin changes for the I2C bus.

Thanks!

jellium, 2017-06-06:
Hi @Bucknalla, thanks for your answer.

It effectively works when one replaces pins `P9` and `P10` by pins `P22` and `P21`, respectively (as depicted on the [pinout diagram](https://docs.pycom.io/pycom_esp32/_downloads/pysensePinoutComp.pdf)). 

Can't wait to grab my hands on the official library! (-;

Bucknalla, 2017-06-06:
No problem! Hold tight, should be ready by the end of the week, we're just making some final tweaks/adjustments!

closed 2017-06-06 by Bucknalla