pycom/pycom-libraries #16: Contribution: Basic GPS/NMEA Decoder for PyTrack

aldrinleal, 2017-06-11:
I've had to use the GPS with PyTrack, but since it wasn't ready, I've wrote my own code. 

My only doubt is: SyPi has two I2C Buses, right? Would Pytrack with SyPi issue an error, since using a Bus Id other than zero throws an error?

https://gist.github.com/aldrinleal/09f9416c84e9574f319a2961411f2a5d

If you have any CLA/Paperwork needs, just let me know ok?

Thank you

livius2, 2017-06-28:
your parser is really minimalistic and do not check crc and fix
the best i have found it this
https://github.com/inmcm/micropyGPS/blob/master/micropyGPS.py

it need some modifications and extend error handling.
Also at the end you need to add GLONASS comands to the list

Cinezaster, 2017-08-17:
Any updates on a fully implemented NMEA decoder? 
I'm going to build a ttn-mapper (maps things network gateways) but I also need altitude and hdop.

mgranberryto, 2017-08-27:
The fields that ttn-mapper needs are all in the GPGGA sentence.  Parsing it is trivial.

Cinezaster, 2017-08-28:
@mgranberryto I know. 
I already implemented xxGGA, xxRMC and xxGSA decoding.

fusspawn, 2018-12-17:
@Cinezaster  Did you manage to finish your TTN mapper implementation. If you'd be willing to share some of our internal TTN fans (Ie me.) would love to see it :)