pycom/pycom-libraries #18: Parsing of GPRMC messages

cfaessler, 2017-06-19:
Currently only GLL messages (only containing location information) are parsed. It would be useful if parsing of GPRMC messages could be done also including speed and direction information

Bucknalla, 2017-06-19:
Hi @cfaessler this is coming in a version 1.1.0 update! Please bear with us as it is on it's way!

Thanks!

rolandvs, 2017-11-27:
@Bucknalla Has the update surfaced yet (since jul 2017)?

Helmond, 2019-03-26:
is there any news on this update?

cfaessler, 2019-03-26:
@Helmond this looks like it: https://github.com/pycom/pycom-libraries/commit/dcaf12bfce3d881d011bfeb038f0bf3c078c8714 but I havent run it

Helmond, 2019-03-26:
> @Helmond this looks like it: [dcaf12b](https://github.com/pycom/pycom-libraries/commit/dcaf12bfce3d881d011bfeb038f0bf3c078c8714) but I havent run it

no that is solving an other bug. it stull doesn't read the speed and direction

cfaessler, 2019-03-26:
> > @Helmond this looks like it: [dcaf12b](https://github.com/pycom/pycom-libraries/commit/dcaf12bfce3d881d011bfeb038f0bf3c078c8714) but I havent run it
> 
> no that is solving an other bug. it still doesn't read the speed and direction
right

nicoduck, 2020-09-06:
Getting speed and optionally height and direction would be a great addition for mapping and other use cases.