pycom/pycom-libraries #19: LoRA OTAA nano gateway example error

rafi021, 2017-06-29:
I have imported the exact code(OTAA node) given in lorawan nano gateway.  But this code gives an ValueError: invalid argument(s) module on this given line. My Lopy module is running latest firmware 1.7.5.b1

lora.add_channel(0, frequency=915000000, dr_min=0, dr_max=5)

Please need help on this. What wrong i have made.

jmarcelino, 2017-06-29:
For 915Mhz LoRaWAN regions the maximum dr (Data Rate index) is 4, so please use 
lora.add_channel(0, frequency=915000000, dr_min=0, dr_max=4)

rafi021, 2017-06-29:
hi @jmarcelino,  thanks. it solve my problem. but otaa node does not get connected to ttn gateway(another lopy). it shows only not joined yet message  all the time. I have edited the code with my device dev_eui,app_eui and app_key. but still no luck. likely my ttn gateway is connected with thingsnetwork. will you please help me to figure out where did I make mistake.  should I post my code here?

rafi021, 2017-06-29:
``""" OTAA Node example compatible with the LoPy Nano Gateway """

from network import LoRa
import socket
import binascii
import struct
import time

# Initialize LoRa in LORAWAN mode.
lora = LoRa(mode=LoRa.LORAWAN)

# create an OTA authentication params
dev_eui = binascii.unhexlify('00 E2 44 46 7D 5E 4A C0'.replace(' ',''))
app_eui = binascii.unhexlify('70 B3 D5 7E F0 00 5F F6'.replace(' ',''))
app_key = binascii.unhexlify('4B CD E4 AA A8 A6 14 13 4A BE 51 D8 38 AE 45 94'.replace(' ',''))

# set the 3 default channels to the same frequency (must be before sending the OTAA join request)
lora.add_channel(0, frequency=915100000, dr_min=0, dr_max=4) #9232 00000
lora.add_channel(1, frequency=915100000, dr_min=0, dr_max=4)
lora.add_channel(2, frequency=915100000, dr_min=0, dr_max=4)

# join a network using OTAA
lora.join(activation=LoRa.OTAA, auth=(dev_eui, app_eui, app_key), timeout=0)

# wait until the module has joined the network
while not lora.has_joined():
    time.sleep(2.5)
    print('Not joined yet...')

# remove all the non-default channels
for i in range(3, 16):
    lora.remove_channel(i)

# create a LoRa socket
s = socket.socket(socket.AF_LORA, socket.SOCK_RAW)

# set the LoRaWAN data rate
s.setsockopt(socket.SOL_LORA, socket.SO_DR, 4)

# make the socket blocking
s.setblocking(False)

time.sleep(5.0)

for i in range (200):
    s.send(b'PKT #' + bytes([i]))
    time.sleep(4)
    rx = s.recv(256)
    if rx:
        print(rx)
    time.sleep(6)

jmarcelino, 2017-06-29:
Yes that code won't work for the US915Mhz region, it's meant for Europe 868 where there are only 3 default channels.

Please use this helpful guide by a community member on how to connect both your nano gateway and node properly. Use the code he links to:

https://seasluglabs.io/blogs/guides/the-things-network-us-setup-with-lopy-as-a-nano-gateway-and-lopy-node

Bucknalla, 2017-07-18:
Thanks for supporting this @jmarcelino, I'll get the seaslug tutorial up onto our own documentation in the near future as well.