pycom/pycom-libraries #21: pytrack: loses gps fix then crashes

ozeta, 2017-07-17:
wipy2.0 

os.uname()=

(sysname='WiPy', nodename='WiPy', release='1.7.5.b2', version='v1.8.6-694-g25826866 on 2017-06-29', machine='WiPy with E
SP32')

pytrack dfu ver: 0.0.3

1) loaded pytrack.py,L76GNSS.py and other classes from pycom github repo into /lib
2) created the attached code (inside zip archive):
[gpsNtpSaveTest_cpy.zip](https://github.com/pycom/pycom-micropython-sigfox/files/1119810/gpsNtpSaveTest_cpy.zip)

after some seconds of getting cordinates into the while loop, l76.coordinates() starts alternating (None,None)
and the correct values.
[image]

after another some seconds the device rebooted by itself
[image]

jmarcelino, 2017-07-17:
Have you tried the new PyTrack firmware (0.0.4)? Sounds like it could help https://forum.pycom.io/topic/1497/new-pytrack-and-pysense-firmware-release-0-0-4

ozeta, 2017-07-17:
I didn't get the new on the forum. I'll install and try it asap and I'll let you know!

ozeta, 2017-07-17:
Just tested, it keeps crashing:

the first exception is this:

```
Guru Meditation Error of type LoadProhibited occurred on core  0. Exception was unhandled.                              
Register dump:                                                                                                          
PC      : 0x40086b35  PS      : 0x00060033  A0      : 0x400880bb  A1      : 0x3ffc5950                                  
A2      : 0x00050023  A3      : 0x00000000  A4      : 0x00000001  A5      : 0x3ffc3cc8                                  
A6      : 0x3ff000dc  A7      : 0x00000001  A8      : 0x00000005  A9      : 0x00000001                                  
A10     : 0x3ffc3d34  A11     : 0x00000000  A12     : 0x3ffc3d34  A13     : 0x00000000                                  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000011  EXCCAUSE: 0x0000001c                                  
EXCVADDR: 0x00000048  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000                                  
                                                                                                                        
Backtrace: 0x40086b35:0x3ffc5950 0x400880bb:0x3ffc5970 0x40088075:0x4000bff0         
```  

then it keeps trowing exception and the backtrace keeps getting bigger:

this is after some throwing

``` 
Guru Meditation Error of type LoadProhibited occurred on core  0. Exception was unhandled.                              
Register dump:                                                                                                          
PC      : 0x40088b1a  PS      : 0x00060033  A0      : 0x80087b1c  A1      : 0x3ffc4e10                                  
A2      : 0x3ffc4e60  A3      : 0x3ffc4e30  A4      : 0x00000020  A5      : 0x3ffc3d2c                                  
A6      : 0x3ffc54d0  A7      : 0x3ffc53f0  A8      : 0x00000000  A9      : 0x00000000                                  
A10     : 0x3ffc4e60  A11     : 0x3ffc3d34  A12     : 0x00000000  A13     : 0x3ffc51f0                                  
A14     : 0x3ffc5220  A15     : 0x00000020  SAR     : 0x0000001a  EXCCAUSE: 0x0000001c                                  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000                                  
                                                                                                                        
Backtrace: 0x40088b1a:0x3ffc4e10 0x40087b1c:0x3ffc4e30 0x400893a4:0x3ffc4e60 0x40089719:0x3ffc5030 0x40089084:0x3ffc5070
 0x400891db:0x3ffc50f0 0x40081535:0x3ffc5110 0x40088b1a:0x3ffc51d0 0x40088b1a:0x3ffc51f0 0x400893a4:0x3ffc5220 0x4008971
9:0x3ffc53f0 0x40089084:0x3ffc5430 0x400891db:0x3ffc54b0 0x40081535:0x3ffc54d0 0x40088b1a:0x3ffc5590 0x40088b1a:0x3ffc55
b0 0x400893a4:0x3ffc55e0 0x40089719:0x3ffc57b0 0x40089084:0x3ffc57f0 0x400891db:0x3ffc5870 0x40081535:0x3ffc5890 0x40086
b35:0x3ffc5950 0x40086b35:0x3ffc5970 0x40088075:0x4000bff0                                                             
```  

at last, only the backtrace is printed:
```  
Backtrace: 0x40088ade:0x3ffc3010 0x40087b1c:0x3ffc3030 0x400893a4:0x3ffc3060 0x40089719:0x3ffc3230 0x40089084:0x3ffc3270
 0x400891db:0x3ffc32f0 0x40081535:0x3ffc3310 0x40088ade:0x3ffc33d0 0x40088ade:0x3ffc33f0 0x400893a4:0x3ffc3420 0x4008971
9:0x3ffc35f0 0x40089084:0x3ffc3630 0x400891db:0x3ffc36b0 0x40081535:0x3ffc36d0 0x40088ade:0x3ffc3790 0x40088ade:0x3ffc37
b0 0x400893a4:0x3ffc37e0 0x40089719:0x3ffc39b0 0x40089084:0x3ffc39f0 0x400891db:0x3ffc3a70 0x40081535:0x3ffc3a90 0x40088
ade:0x3ffc3b50 0x40088ade:0x3ffc3b70 0x400893a4:0x3ffc3ba0 0x40089719:0x3ffc3d70 0x40089084:0x3ffc3db0 0x400891db:0x3ffc
3e30 0x40081535:0x3ffc3e50 0x40088b1a:0x3ffc3f10 0x40088b1a:0x3ffc3f30 0x400893a4:0x3ffc3f60 0x40089719:0x3ffc4130 0x400
89084:0x3ffc4170 0x400891db:0x3ffc41f0 0x40081535:0x3ffc4210 0x40088b1a:0x3ffc42d0 0x40088b1a:0x3ffc42f0 0x400893a4:0x3f
fc4320 0x40089719:0x3ffc44f0 0x40089084:0x3ffc4530 0x400891db:0x3ffc45b0 0x40081535:0x3ffc45d0 0x40088b1a:0x3ffc4690 0x4
0088b1a:0x3ffc46b0 0x400893a4:0x3ffc46e0 0x40089719:0x3ffc48b0 0x40089084:0x3ffc48f0 0x400891db:0x3ffc4970 0x40081535:0x
3ffc4990 0x40088b1a:0x3ffc4a50 0x40088b1a:0x3ffc4a70         
```  

and finally, it reboots:

```  
uru Meditation Error of type Guru Meditation Error of type Guru Meditation Error of type Guru Med�@P)�?Error of typ�@3@
 0�@3��40088cad��3ff81fd0@@ 0�@3��4008147b��3fffffe4@@ 0�@3@ 0�@3��40088caa��3fffffe4@0@d-�?00000000�%�?d-�?3ff82044$6�?
ets Jun  8 2016 00:22:57                                                                                                
                                                                                                                        
rst:0x9 (RTCWDT_SYS_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)                                                              
configsip: 0, SPIWP:0x00                                                                                                
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00                                                 
mode:QIO, clock div:2                                                                                                   
load:0x3fff9010,len:12                                                                                                  
[... cut ...]
load:0x40078000,len:11412                                                                                               
load:0x4009fc00,len:848                                                                                                 
entry 0x4009fd9c                                                                                                        
MicroPython v1.8.6-694-g25826866 on 2017-06-29; WiPy with ESP32                                                         
Type "help()" for more information.                                     
```

jmarcelino, 2017-07-17:
Not sure if it help but can't hurt to update the firmware on the WiPy as well, version 1.7.6b1 is available
https://forum.pycom.io/topic/1457/new-firmware-release-1-7-6-b1

ozeta, 2017-07-18:
just installed new firmware, is there a way to copy paste the repl full output?

pytrack is now more stable. before 1.7.6b1 it started getting no coords, then crashed after 100-120 cycles.
now it has run for for 1000 cycles more or less, however, after the first 100, it started alternating good reading with none.

I changed a bit the script:

```
while True:
    print("i: {} loc: {} memfree: {}".format(i, l76.coordinates(), gc.mem_free()/1000))
    i += 1

```
After starting, the memfree is 45.

output at the end:
```

i: 1010 loc: (40.80619, 14.33766) memfree: 3.68
i: 1011 loc: (None, None) memfree: 4.4
i: 1012 loc: (None, None) memfree: 4.64
i: 1013 loc: (None, None) memfree: 4.48
i: 1014 loc: (40.80619, 14.33766) memfree: 3.44
i: 1015 loc: (None, None) memfree: 4.352
i: 1016 loc: (None, None) memfree: 4.512
i: 1017 loc: (None, None) memfree: 4.384
i: 1018 loc: (None, None) memfree: 4.352
i: 1019 loc: (None, None) memfree: 4.288
i: 1020 loc: (None, None) memfree: 4.464
i: 1021 loc: (None, None) memfree: 4.304
i: 1022 loc: (40.80619, 14.33766) memfree: 3.248
i: 1023 loc: (None, None) memfree: 4.24
Traceback (most recent call last):
  File "<stdin>", line 12, in <module>
  File "/flash/lib/L76GNSS.py", line 42, in coordinates
MemoryError: memory allocation failed, allocating 8192 bytes

```

Bucknalla, 2017-07-18:
Looking into this now! Will report back with what I find! Thanks :)

Bucknalla, 2017-07-19:
Hey @ozeta, could you please try this GPS file (https://github.com/pycom/pycom-libraries/blob/develop/pytrack/lib/L76GNSS.py)? I tracked the memory leak down to the alarm function in the GPS lib.

I'm now taking a look at the C lib for the alarm and trying to work out why the alarm objects aren't being released from the heap.

Please let me know if this works for you!

Thanks!

Alex

ozeta, 2017-07-19:
it works really good!

```

772 (40.80623, 14.33749). memfree: 4.5504
773 (40.80623, 14.33749). memfree: 4.5120
774 (40.80623, 14.33749). memfree: 4.6112
775 (40.80623, 14.33749). memfree: 4.5184
```

thanks!

Bucknalla, 2017-07-19:
Great! I'll merge this into the master and close this issue :)

closed 2018-07-05 by sdaniel55