pycom/pycom-libraries #23: calibrate_rtc error after a go_to_sleep ( pytrack )

jiemde, 2017-07-20:
I have an error after an "go_to_sleep"
5 pas de coordonnees GPS
go to deepsleep
Traceback (most recent call last):
  File "<stdin>", line 85, in <module>
  File "/flash/lib/pytrack.py", line 134, in setup_sleep
  File "/flash/lib/pytrack.py", line 157, in calibrate_rtc
IndexError: list index out of range
>
MicroPython v1.8.6-694-g25826866 on 2017-06-29; SiPy with ESP32
Type "help()" for more information.
>>>
>>>

jiemde, 2017-07-20:
everything is up to date ( firmware for the SiPy, for the pytrack, last pytrack.py lib )

danicampora, 2017-07-21:
Should be solved with this: https://github.com/pycom/pycom-libraries/commit/0ac31b6e38f75359ddd5fd6288a1b5de75b2e33b

closed 2017-07-21 by danicampora

jiemde, 2017-07-21:
> Should be solved with this: 0ac31b6

Many thanks Daniel it's work!

Jiemde

<https://mailtrack.io/> Sent with Mailtrack
<https://mailtrack.io/install?source=signature&lang=en&referral=jmdefraine@gmail.com&idSignature=22>

Jean-Marie Defraine / Project Manager
+32 (0)497043637 <+32%20497%2004%2036%2037>/
defraine.jm@digital-media-consulting.be

Digital Media Consulting sprl Office: +32 (0)4/3720716
<+32%204%20372%2007%2016> / Fax: +32 (0)4/3720719 <+32%204%20372%2007%2019>
Rue Jean Jaurès, 176 - 4430 ANS - BELGIUM
http://www.digital-media-consulting.be

[image: Facebook] <http://www.facebook.com/DynamicDigitalSign> [image:
Twitter] <http://twitter.com/DynDigitalSign> [image: Google Plus]
<http://defraine.jm@gmail.com/> [image: Skype] <http://jmdgtd/>

This e-mail message may contain confidential or legally privileged
information and is intended only for the use of the intended recipient(s).
Any unauthorized disclosure, dissemination, distribution, copying or the
taking of any action in reliance on the information herein is prohibited.
E-mails are not secure and cannot be guaranteed to be error free as they
can be intercepted, amended, or contain viruses. Anyone who communicates
with us by e-mail is deemed to have accepted these risks. Company Name is
not responsible for errors or omissions in this message and denies any
responsibility for any damage arising from the use of e-mail. Any opinion
and other statement contained in this message and any attachment are solely
those of the author and do not necessarily represent those of the company.