pycom/pycom-libraries #24: I2C Error pysense board

rolandvs, 2017-07-26:
Dear,

I have a `loPy (1.7.7.b1)` with a `Pysense` board. I changed the `sense.py` example into `main.py` with a few changes. Nothing special. However after running it, it will crash/stop.

```
Traceback (most recent call last):
File "main.py", line 41 in <module>
File "main.py", line 35 in <main>
File "/flash/lib/pysense.py", line 177, in read_battery_voltage
File "/flash/lib/pysense.py", line 118, in peek_memory
File "/flash/lib/pysense.py", line 85, in _write
File "/flash/lib/pysense.py", line 97, in _wait
OSError: I2C bus error
Micropython v1.8.6-715-g5fa16c2c on 2017-07-20; LoPy with ESP32
```
There is no I2C error but reading (`peeking`) the battery voltage is generating the problem.

Rebooting it by ctrl-C, ctrl-D let it tells me that the `Pysense` board cannot be found. Looking at the code the problem seems to be in the __init__ part and the `read_battery_voltage` part of the code. Disabling the ADC init code and not using the py_read_battery_voltage() anymore keeps everything running.

Only de-powering the unit completely (no battery/no USB) will "recognise" the board again, but any use of the read_battery_voltage will raise an exception again.

danicampora, 2017-07-26:
Hi,

When you read the firmware version of your Pysense, what value do you get?

rolandvs, 2017-07-26:
pysense.__version__ reports 1.0.2

danicampora, 2017-07-26:
@rolandvs that's not what I meant... Can you call the `.read_fw_version()` from the pysense library and see what it returns? I think you have an old firmware on your Pysense...

rolandvs, 2017-07-26:
The `read_fw-version` returned `1` and I reprogrammed the `pysense` and `pytrack` boards to version `0.0.4`. The code seems to be working now as expected. Thanks!

(in the mean time I almost finished decoding $GNMRC instead of $GNGLL to get date/time to set the RTC when GPS is available).

closed 2017-07-26 by rolandvs