pycom/pycom-libraries #26: DeepSleep library example do not work

txinto, 2017-08-16:
Hello, I've tried to execute the example of DeepSleep.

I followed this guide, trying to figure the obscure steps:
https://docs.pycom.io/chapter/datasheets/boards/deepsleep/api.html

First, I added this example as a project in Atom:
https://github.com/pycom/pycom-libraries/tree/master/examples/deepsleep
So, I have a single main.py on my project.

Then I've copied the deepsleep.py file to the the ./lib/deepsleep/deepsleep.py path in my project.
The file I've copied is this: https://github.com/pycom/pycom-libraries/blob/master/deepsleep/deepsleep.py
Now I have two files
./main.py
./lib
./lib/deepsleep/deepsleep.py

Then I sync the project with my SiPy using the Atom Pimakr plugin and after the reset I see this:
>>> Traceback... 
  File "<stdin>", line 3, in <module>
TypeError: 'module' object is not callable
>

I've tried to change 
ds = DeepSleep()
to
ds = DeepSleep.DeepSleep()

and then 

Traceback (most recent call last):
  File "main.py", line 6, in <module>
  File "/flash/lib/deepsleep/DeepSleep.py", line 145, in get_wake_status
  File "/flash/lib/deepsleep/DeepSleep.py", line 76, in peek
IndexError: bytes index out of range

Thanks for fixing the documentation and/or example and/or library.

Best regards.

txinto, 2017-08-16:
I've tried to move deepsleep.py to the root folder and sync:

Traceback (most recent call last):
  File "main.py", line 6, in <module>
  File "deepsleep.py", line 145, in get_wake_status
  File "deepsleep.py", line 76, in peek
IndexError: bytes index out of range

jhoguaman, 2017-10-24:
Hi, I had the same problem, but I was working whit supply voltaje 3.3 V. You must to change to 5V the supply voltaje. According to the schematic:

https://docs.pycom.io/chapter/datasheets/downloads/deepsleep-schematic.pdf

emartinez3, 2017-12-11:
Most recent version of code, latest firmware and I am also seeing the exact same problem.

Traceback (most recent call last):
File "main.py", line 6, in 
File "deepsleep.py", line 145, in get_wake_status
File "deepsleep.py", line 76, in peek
IndexError: bytes index out of range

I am powered via the USB connector.

nixman2, 2018-04-26:
same here.

(sysname='WiPy', nodename='WiPy', release='1.17.3.b1', version='v1.8.6-849-83e2f7f on 2018-03-19', machine='WiPy with ESP32')

on a WiPy 3 board

`♦Traceback (most recent call last):
  File "<stdin>", line 22, in <module>
  File "/flash/lib/deepsleep.py", line 157, in get_wake_status
  File "/flash/lib/deepsleep.py", line 81, in peek
IndexError: bytes index out of range`

I am powered through the USB connector.

Xykon, 2018-04-27:
This library is only used with the Deep Sleep shield. https://docs.pycom.io/chapter/datasheets/boards/deepsleep.html

The WiPy3 has its own deep sleep functionality, which is part of the machine module https://docs.pycom.io/chapter/firmwareapi/pycom/machine/

sdaniel55, 2018-07-05:
Hi @txinto, you should have 2 files:
```
- main.py
- lib/
    |- deepsleep.py
```

Libraries always go into the `lib/` folder. Then you can import and use `DeepSleep` in `main.py`.

I didn't manage to reproduce the issue, so if anyone could provide example code, I can try again. (@jhoguaman, @wink15987, @nixman2)

vinnie38170, 2018-12-10:
On a Sipy with a pytrack :

// Dosomething
ds = DeepSleep()
ds.go_to_sleep(10)  # go to sleep for 10 seconds
// do something

It Never wakes up.
(I have the added shield corrector placed under the sipy)

Please help.

vinnie38170, 2018-12-10:
If I'm not supposed to use go_to_sleep() with the shield, how do I put the sipy to deepsleep for a specific amount of time?

skylermaxwell42, 2019-01-19:
Has there been any progress on this? I am running the latest version of the firmware on the WiPy 3.0 and I am still having the same issue. Both the external lib and the builtin deep sleep libraries won't work.

vinnie38170, 2019-01-21:
I actually stopped using this one and put to DeepSleep my Pytrack instead. It cuts the whole thing and saving much more energy