pycom/pycom-libraries #27: LoRaWAN Nano Gateway Crashes

aizukanne, 2017-08-19:
Hi,

I implemented the LoRaWan  nanogateway. It works but keeps crashing and restarting at intervals with the following error message after what looks like a long hex dump.

`Guru Meditation Error of type LoadProhibited occurred on core  0. Exception was unhandled.
Register dump:
PC      : 0x400879fd  PS      : 0x00060033  A0      : 0x80088f97  A1      : 0x3ffe37a0
A2      : 0x3ffe3800  A3      : 0x3ffe37d0  A4      : 0x00000020  A5      : 0x3ffc48b8
A6      : 0x00001182  A7      : 0x3ffe3da0  A8      : 0x3ff40000  A9      : 0x0000007d
A10     : 0x00000002  A11     : 0x3ffe3bb0  A12     : 0x8008b2f8  A13     : 0x3ffe3ba0
A14     : 0x3ffe3bd0  A15     : 0x00000020  SAR     : 0x00000012  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00001186  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000

Backtrace: 0x400879f▒ets Jun  8 2016 00:22:57`

Any help solving this would be appreciated.

A. Izukanne

danicampora, 2017-08-20:
Hello @aizukanne what firmware version are you running on the LoPy?

aizukanne, 2017-08-20:
Hello,
Firmware version is 1.7.9.b3
Terminal shows MicroPython v1.8.6-747-gc4ec6110.

aizukanne, 2017-08-23:
Hello @danicampora 

When I decoded the exception, this is what I got

`Decoding 51 results
0x400879fd: bt_i2c_write_set at ?? line ?
0x40088f94: ram_gen_rx_gain_table at ?? line ?
0x4008b2f5: rf_cal_data_backup at ?? line ?
0x4008b66a: phy_init_pwr_print at ?? line ?
0x4008af75: register_chipv7_phy_init_param at ?? line ?
0x4008b0f0: phy_byte_to_word at ?? line ?
0x4008168a: esp_intr_disable at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./intr_alloc.c line 626
0x400879fa: bt_i2c_write_set at ?? line ?
0x400879fa: bt_i2c_write_set at ?? line ?
0x4008b2f5: rf_cal_data_backup at ?? line ?
0x4008b66a: phy_init_pwr_print at ?? line ?
0x4008af75: register_chipv7_phy_init_param at ?? line ?
0x4008b0f0: phy_byte_to_word at ?? line ?
0x4008168a: esp_intr_disable at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./intr_alloc.c line 626
0x400879fa: bt_i2c_write_set at ?? line ?
0x400879fa: bt_i2c_write_set at ?? line ?
0x4008b2f5: rf_cal_data_backup at ?? line ?
0x4008b66a: phy_init_pwr_print at ?? line ?
0x4008af75: register_chipv7_phy_init_param at ?? line ?
0x4008b0f0: phy_byte_to_word at ?? line ?
0x4008168a: esp_intr_disable at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./intr_alloc.c line 626
0x400879fa: bt_i2c_write_set at ?? line ?
0x400879fa: bt_i2c_write_set at ?? line ?
0x4008b2f5: rf_cal_data_backup at ?? line ?
0x4008b66a: phy_init_pwr_print at ?? line ?
0x4008af75: register_chipv7_phy_init_param at ?? line ?
0x4008b0f0: phy_byte_to_word at ?? line ?
0x4008168a: esp_intr_disable at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./intr_alloc.c line 626
0x400879fa: bt_i2c_write_set at ?? line ?
0x400879fa: bt_i2c_write_set at ?? line ?
0x4008b2f5: rf_cal_data_backup at ?? line ?
0x4008b66a: phy_init_pwr_print at ?? line ?
0x4008af75: register_chipv7_phy_init_param at ?? line ?
0x4008b0f0: phy_byte_to_word at ?? line ?
0x4008168a: esp_intr_disable at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./intr_alloc.c line 626
0x400879fa: bt_i2c_write_set at ?? line ?
0x400879fa: bt_i2c_write_set at ?? line ?
0x4008b2f5: rf_cal_data_backup at ?? line ?
0x4008b66a: phy_init_pwr_print at ?? line ?
0x4008af75: register_chipv7_phy_init_param at ?? line ?
0x4008b0f0: phy_byte_to_word at ?? line ?
0x4008168a: esp_intr_disable at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./intr_alloc.c line 626
0x400879fa: bt_i2c_write_set at ?? line ?
0x400879fa: bt_i2c_write_set at ?? line ?
0x4008b2f5: rf_cal_data_backup at ?? line ?
0x4008b66a: phy_init_pwr_print at ?? line ?
0x4008af75: register_chipv7_phy_init_param at ?? line ?
0x4008b0f0: phy_byte_to_word at ?? line ?
0x4008168a: esp_intr_disable at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./intr_alloc.c line 626
0x400879fa: bt_i2c_write_set at ?? line ?
0x400879fa: bt_i2c_write_set at ?? line ?
`

danicampora, 2017-08-23:
@aizukanne are you using a LoPy OEM?

aizukanne, 2017-08-23:
@danicampora No I am not. I installed on an ESP32 with RFM95 connected as shown in the the LoPy PINOUT diagram.

danicampora, 2017-08-23:
OK I see... The crash reports shows that the issues are in the WiFi and BT Phy functions. Maybe a power supply problem, or an antenna matching issue.

aizukanne, 2017-08-23:
@danicampora Thanks.
I am using the built-in antenna. Power is from USB. Is it possible to disable BT?

caseyrodgers, 2018-02-08:
Did you ever find a solution to this?  I'm having the same issue.  

I'm using a LoPy4 with firmware 1.15.0.b1.

aizukanne, 2018-02-09:
Yes, I found out the script I was using was incomplete. I downloaded a more recent version and everything or almost everything worked like a charm.

I said almost cos I couldn't get OTAA to work, probably cos I had a different LoRa model.

sdaniel55, 2018-07-05:
Hi @caseyrodgers, I'll close this issue, but if you're still having problems, feel free to reopen it and add more details. Thanks!

closed 2018-07-05 by sdaniel55