pycom/pycom-libraries #35: MemoryError: memory allocation failed,

fableh, 2017-10-27:
Hi guys,

iam using the LoPy board to try out the "Pytrack" sample but i got the following error:

Traceback (most recent call last):
  File "main.py", line 35, in <module>
  File "/flash/lib/L76GNSS.py", line 55, in coordinates
MemoryError: memory allocation failed, allocating 3751 bytes
MicroPython v1.8.6-796-g489fafa0 on 2017-10-15; LoPy with ESP32
Type "help()" for more information.

os.uname()=
(sysname='LoPy', nodename='LoPy', release='1.9.2.b2', version='v1.8.6-796-g489fafa0 on 2017-10-15', machine='LoPy with ESP32', lorawan='1.0.0')

pytrack_0.0.6.dfu

what i´am doing wrong here?

thx and br,
fabian

rpcme, 2017-11-09:
+1, received the same error. I think this happens when reading is done from indoors.  When I scaled down the code to give it breathing room, it timed out before hitting this problem.

bogd, 2017-12-02:
+1 here. Exactly the same - after just a few reads (sometimes just one - , the program crashes with MemoryError.

```
RTC Set from NTP to UTC: (1970, 1, 1, 0, 3, 19, 594962, None)
Adjusted from UTC to EST timezone (1970, 1, 1, 2, 3, 19, 3, 1)

(None, None) - (1970, 1, 1, 0, 3, 49, 703270, None) - 37728
Traceback (most recent call last):
  File "<stdin>", line 36, in <module>
  File "/flash/lib/L76GNSS.py", line 56, in coordinates
MemoryError: memory allocation failed, allocating 4064 bytes
```

I'm using PyTrack firmware 0.0.7 (just updated it :) ).

@rpcme - what do you mean when you say "I scaled down the code"?

PhilipOlsson, 2017-12-09:
+1,
I also see the same issue with just the minimal pytrack example code on:

>>> os.uname()
(sysname='LoPy', nodename='LoPy', release='1.10.2.b1', version='v1.8.6-849-g0af003a4 on 2017-11-24', machine='LoPy with ESP32', lorawan='1.0.0')

Pytrack 0.0.7

EdsonAlcala, 2018-03-27:
any news about this issue?

danicampora, 2018-03-27:
This should have been fixed long ago with this line: https://github.com/pycom/pycom-libraries/blob/master/pytrack/lib/L76GNSS.py#L75

Is that not the case? Are you still getting the memory allocation error?

bogd, 2018-04-12:
I was able to get a GPS fix after changing the nmea length as per the above fix. However, I do remember having to manually edit the file in order to make that change. Is it possible that this was before the fix made it into the official library?

Anyway, as far as I can tell, the issue is now solved. Thank you!

closed 2018-07-05 by sdaniel55