pycom/pycom-libraries #38: Data not sent to TTN

Glagnar, 2017-11-12:
I have a LoPy that is connected to the TTN. While monitoring the serial in a terminal I also see it receiving data.
Unfortunately none of this data can be seen in the TTN console.
[image]
[image]

jmarcelino, 2017-11-12:
Hi, the first packet on your screenshot is a "Join Accept" message, you shouldn't see these on your nano-gateway since they are usually sent with an inverted IQ (so gateways don't listen to other gateways). Do you know if you have LoRa equipment in the area? Could also be just random transmission from other devices/noise... in any case TTN will definitely drop/ignore it.

The second message is more interesting, it's a Join Request for a node with App EUI 09C9800026051201 and device EUI E477E164DBD6AB19. Are these familiar to you? It's strange because native TTN App EUIs start with prefix 70B3D57E and not 09C980, but it is possible to customize. 

So we can help you better can you describe what is your node(s) and node configuration you are using? 

If you prefer to discuss this in private you can reach me by e-mail at jmarcelino@pycom.io

Thanks.

Glagnar, 2017-11-12:
Thank you for the swift feedback.
I have checked, and the Sodaq board I am trying to connect is this one. You may notice that it was connected to TTN while I was driving around a nearby city - but I live in a small village, and here it does not connect to my LoPy.
[image]

jmarcelino, 2017-11-12:
Thanks. Unfortunately I don't see anything from that device in your screenshot, just that OTAA device so not even ABP as the Sodaq is using.

 Is it possible to force it to transmit while you capture the output of the gateway?

Also remember the LoPy is a is a single channel (nano) gateway, unlike a full gateway it only listens  on one frequency (868.1Mhz) and one spreading factor while your node may be configured for a gateway which listens on 8 frequencies and several spreading factors. 

If it's a RN2483 based Sodaq node I recommend changing your configuration to:

```
mac set ch status 1 off
mac set ch status 2 off
mac set ch status 3 off
mac set ch status 4 off
mac set ch status 5 off
mac set ch status 6 off
mac set ch status 7 off

mac set dr 5
```

to make sure you only use only channel 0 (868.1Mhz) and SF7/125Khz

Glagnar, 2017-11-13:
Thank you again for your input:
I issued the commands, but the 'Mac set ch status 3 ... 7' gave me an error: "invalid_param" but aside that, it hasn't changed the problem that I dont receive data.

I will try and set up an additional LoPy as a node, to see if it is a problem with the 'nano-gateway' or perhaps the Sodaq.

jmarcelino, 2017-11-13:
Ok, that probably means channel 3..7 weren't enabled to start with. 
Did you do the `mac set dr 5`? If the Sodaq is really sending uplinks on 868.1Mhz (default for channel 0) using SF7/125 the LoPy should pick it up.

If you have another LoPy you can use the single channel ABP node example: https://github.com/pycom/pycom-libraries/blob/master/examples/lorawan-nano-gateway/abp_node.py

Just replace the ABP authentication params with your own from TTN.

Glagnar, 2017-11-13:
The LoPy to LoPy works, without me having to do anything on the 'Gateway LoPy'.
I will look more deeply into the Sodaq Explorer.

sdaniel55, 2018-07-05:
Hi @Glagnar, I'll close this issue for now. If you're still having problems, feel free to reopen it and add more details.

closed 2018-07-05 by sdaniel55

mhmtkrl, 2018-09-20:
Hi,
I have a LoPy nano Gateway that is connected to the TTN. In a terminal, I can see receiving data.
But I can't see data in the TTN application data console. I am trying to communicate between STM32 LoRa Module and LoPy nano Gateway using OTAA. 
Could you please help me?
Here is my screenshots.
[image]
[image]

robert-hh, 2018-09-20:
Lookign at the  gateway traffic if is strange, that the dev EUI changes for every message. I would expect it to be the same all the time.

mhmtkrl, 2018-09-20:
Yes, I don't understand why it is doing that. I can not connect the TTN network but as you can see from the screenshots, the gateway(LoRa nano GW LoPy) is receiving some data.

robert-hh, 2018-09-20:
The APP eui is also changing constantly. You can decode the data in the log of the nanogateway. T's base64 encoded. It contains APP eui and device eui in clear. That should match what you see in the Gateway log.
Do APP EUI and device EUI you see  match anything you have coded in your  STM32 Lora module? That obe should always use the same APP EUI and device EUI.  If not, then you are receiving other device, or your STM32 module runs weird and sends wrong formatted messages.
An rssi of -76db is also pretty low for a situation, where gateway and module are close to each other. and have appropriate antennas.