pycom/pycom-libraries #40: Exception: Pysense board not detected

titusece, 2017-11-13:
Hello,

I have connected the pycom wipy 2.0 board into pysense board and updated this libraries into it.
Now getting the following errors.
Can you please tell me what could be the problem ?

```
>>> ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff9028,len:8
load:0x3fff9030,len:932
load:0x4009fa00,len:0
ho 12 tail 0 room 4
load:0x4009fa00,len:13232
entry 0x400a04e8
1
Traceback (most recent call last):
  File "main.py", line 9, in <module>
  File "/flash/lib/pysense.py", line 102, in __init__
Exception: Pysense board not detected
MicroPython v1.8.6-820-g56004e13 on 2017-11-08; WiPy with ESP32
Type "help()" for more information.
>>>
>>>
```

jmarcelino, 2017-11-14:
Have you tried updating the firmware of the Pysense? Like this https://docs.pycom.io/chapter/pytrackpysense/installation/firmware.html

titusece, 2017-11-14:
Thanks for your reply. I have installed the FW using below command in WIN7.
dfu-util-static.exe -D pysense_0.0.7.dfu

Now, I can't see any USB COM port detected in system but 'libusbK USB Devices' & Pysense.
Can you please help me now ?

titusece, 2017-11-15:
Hi jmarcelino,
Can you please help me further ?

titusece, 2017-11-17:
Uninstalled the previously installed driver and installed the below driver, then it worked.
https://docs.pycom.io/chapter/pytrackpysense/installation/drivers.html
https://docs.pycom.io/chapter/pytrackpysense/installation/pycom.inf

closed 2017-11-17 by titusece