pycom/pycom-libraries #42: TypeError: object with buffer protocol required in accelerometer_wake example

titusece, 2017-11-17:
Hello,
Using 'Pysense' board and getting the below error when I run the **_accelerometer_wake_** example.

```
Traceback (most recent call last):
  File "main.py", line 18, in <module>
  File "/flash/lib/LIS2HH12.py", line 123, in enable_activity_interrupt
TypeError: object with buffer protocol required
```

https://github.com/pycom/pycom-libraries/blob/master/examples/accelerometer_wake/main.py

What could be the problem ?
Can you please help me ?

jmsalomr, 2018-06-17:
Hello,
Have the same problem.
Any advice?
Can anyone help?

sdaniel55, 2018-07-05:
Hi @titusece and @jmsalomr,

I've just tested the example code, it seems to be working fine. Make sure you're using the latest [firmware](https://docs.pycom.io/chapter/gettingstarted/installation/firmwaretool.html) and [libraries](https://github.com/pycom/pycom-libraries). This might have been fixed in one of the updates. Feel free to reopen the issue if you're still having problems, but I'll close this for now.

closed 2018-07-05 by sdaniel55