pycom/pycom-libraries #46: Pysense MPL3115A2 hangs rather quickly and consistently.

stroobandt, 2017-12-25:
This Pysense MPL3115A2 bug has been [confirmed independently by two forum users](https://forum.pycom.io/topic/2322/mpl3115a2-hangs-intermittently/), myself included.

Within a minute or two (it varies) below script hangs without any error message.

Serial communication also breaks down and a power cycle is required to regain communication with the LoPy Pysense combo.

    import pycom
    import pysense
    import machine
    from MPL3115A2 import MPL3115A2, PRESSURE
    import utime
    import sys

    class Device:

        def __init__(self):
            self.iteration = 0
            self.sense(self)
            self.timer = machine.Timer.Alarm(self.sense, 3, periodic=True)

        def sense(self, device):
            # barometer mode, not raw, oversampling 128, minimum time 512 ms
            self.p = MPL3115A2(board, mode=PRESSURE).pressure()    # in Pa 
            self.iteration += 1
            print('%f Pa | iteration %d' % (self.p, self.iteration))
            pycom.rgbled(0x002200)
            utime.sleep(0.1)
            pycom.rgbled(0x000000)

    pycom.wifi_on_boot(False)
    pycom.heartbeat(False)
    board = pysense.Pysense()
    device = Device()

    try:
        while True:
            pass
    except KeyboardInterrupt:
        device.timer.cancel()
        sys.exit()

stroobandt, 2018-01-03:
I do not know if this is related, but this might be helpful. A [similar problem with an MPL3115A2](https://github.com/adafruit/Adafruit_Python_BMP/issues/4) has been reported on a Raspberry Pi.

`i2c i2c-1: transfer setup timed out` was the error given by the Raspberry Pi.

stroobandt, 2018-01-03:
Does the Pysense employ I2C for communicating with the MPL3115A2?
If so, I am pretty sure we are dealing with an I2C problem.

The following reading material hints into this direction:

- http://www.i2c-bus.org/i2c-primer/obscure-problems/
- https://electronics.stackexchange.com/a/29079/24451

Telepresent, 2021-06-30:
Is it solved?

peter-pycom, 2021-06-30:
Yes. Feel free to report back results

peter-pycom, 2021-06-30:
https://github.com/pycom/pycom-libraries/commit/01ced278aec805236d5ebad6ac37d297e24d6388 <-fix

closed 2021-06-30 by peter-pycom