pycom/pycom-libraries #47: lorawan-nano-gateway gateway error code

jemesmain, 2018-01-04:
hello
being using the exemple file stated above the node don't connect in otaa mode using another lopy as a gateway...
it keeps on saying not joined yet. i try the otaa node code with another ttn gateway and it worked well for both apb and otaa activation so i fugured out it would be on the gateway side

on ttn console only join request on the gateway trafic no up or downlink message.
does anybody else have the same problem...

thanks for your answer

jmarcelino, 2018-01-05:
Since you already did it for the other gateway can you also try an ABP node with your LoPy nanogateway? 

Let us know the result. Thanks!

jemesmain, 2018-01-05:
The APB activation is working fine since the very first time with my 
lopy gateway.

Sorry if I didn't mention it or if that testing was not explicitely 
mentionned by me.

Thanks for your support.

... mais la mise en place de ce qui existe déjà!

<jean-eric.mesmain@laposte.net>
Mobile   : 06 63 95 71 56
----------------------------------------------------
111 avenue la BRUYERE
38100 GRENOBLE
----------------------------------------------------
<http://www.bicyclopresto.fr> Le vélo top chrono
<http://www.energyleaks.org> Solutions intelligentes pour bâtiments Bêtes...
----------------------------------------------------
Jean-Eric MESMAIN

jemesmain, 2018-01-05:
I forget to ask you how I can have more information for troubleshooting 
... whether on the atom pymakr side or on the ttn console side.

Thanks!

-- 
Jean-Eric MESMAIN

L'innovation n'est pas la recherche...
... mais la mise en place de ce qui existe déjà!

<jean-eric.mesmain@laposte.net>
Mobile   : 06 63 95 71 56
----------------------------------------------------
111 avenue la BRUYERE
38100 GRENOBLE
----------------------------------------------------
<http://www.bicyclopresto.fr> Le vélo top chrono
<http://www.energyleaks.org> Solutions intelligentes pour bâtiments Bêtes...
----------------------------------------------------
Jean-Eric MESMAIN

jemesmain, 2018-01-07:
Hello

I started from the very start from lorawan nano gateway

So now the otaa is working but what is bothering me it's that it keeps 
saying not joined yet in the repl pymakr atom screen

I both see traffic on ttn console for gateway and device now...

Before i didn't see any "ansewering message" on gateway traffic ie. now 
i see both green and orange "electric thunder logo"

I suppose the otaa_node program is suppose to "print" the answer of "the 
gateway" but it didn't print

if rx:
         print('Received: {}, on port: {}'.format(rx, port))
     time.sleep(6)

In fact i believe that has not joined yed has stop to print in the repl 
screen we must understand that it has joined ;-)

I modify your program as follow

# wait until the module has joined the network
while not lora.has_joined():
     time.sleep(2.5)
     print('Not joined yet...')

*print('device connected')*

-- 
Jean-Eric MESMAIN

L'innovation n'est pas la recherche...
... mais la mise en place de ce qui existe déjà!

<jean-eric.mesmain@laposte.net>
Mobile   : 06 63 95 71 56
----------------------------------------------------
111 avenue la BRUYERE
38100 GRENOBLE
----------------------------------------------------
<http://www.bicyclopresto.fr> Le vélo top chrono
<http://www.energyleaks.org> Solutions intelligentes pour bâtiments Bêtes...
----------------------------------------------------
Jean-Eric MESMAIN

sdaniel55, 2018-07-05:
Hi @jemesmain, I'll close this issue for now. If you're still having problems, feel free to reopen it and add more details. Thanks!

closed 2018-07-05 by sdaniel55