pycom/pycom-libraries #56: otaa_node never joins

RussNelson, 2018-04-04:
Running the current version of otaa_node.py and it never joins. I've checked and double-checked my eui and key. Gateway is in the other room, and another node is joined to it and sending data. How might I diagnose this problem? Is there any way to find out why the join is failing?

andreac, 2018-04-05:
same problem. Anyone??

AlexandreLogist, 2018-04-30:
Same problem

ryanfeist1, 2018-05-11:
Same here.  Running on original lopy.  Release 1.17.3.b1.  Eventually will join but very inconsistent join times.

mtalhaf, 2018-07-17:
We are currently facing the same issue. We have a LoPy4 v1.18.0 attached to a Expansion board 3 v0.0.9 acting as our LoraWan gateway and a node with a FiPy v1.18.0 attached to a Pytrack v0.0.8.

Everything is set up properly in TTN the gateway shows that its properly connected and the device is correctly configured in the application as well.

We observed that once everything is set up properly, the first time we connect the node to the gateway it joined the network almost immediately and we could see the example messages being sent through. However, upon reconnecting the device we faced the same issue where the node doesn't join the network. 

We wiped everything all the devices and deleted everything from TTN and reinstalled everything from scratch and observed the same thing. It would connect and join the network first time but subsequent attempts would fail.

Investigating this a little further on TTN we can see that the Join request is sent by the device, the gateway accepts the request. But the device never gets the confirmation and keeps trying to join the network in an infinite loop.

Following is a picture of our gateway traffic:
[image]

Attached our code as well.
[PyCom.zip](https://github.com/pycom/pycom-libraries/files/2200881/PyCom.zip)

sdaniel55, 2018-07-24:
Hi everyone 👋 
I was able to reproduce this and looking into this problem. I'll update this issue once I have a solution.

AlexSartori, 2019-05-13:
Same problem for me.

itrich, 2020-06-26:
More than one year later, I'm still facing the same problem...