pycom/pycom-libraries #57: NB-IoT - DNS not working

darox, 2018-04-20:
DNS seems not working on NB-IoT, there's no DNS request in the network trace at all, instead the Fipy sends the TCP packets to various IP-Addresses. Thus MQTT over TLS only works, if the IP and not the FQDN of the broker is specified.

darox, 2018-06-07:
Would be nice to have an answer on this, I could also provide network traces.

darox, 2018-08-26:
Still no answer on this? Your customer support is really bad!

iwahdan88, 2018-08-28:
Hello @darox ,
first, I want to apologise for the big Delay we had responding to your submitted issue, is it possible you could supply me with the firmware version you have on the sequans modem .
you can get this info using the following command:
```
from network import LTE
lte = LTE()
print(lte.send_at_cmd('AT!="showver"'))
```

darox, 2018-08-30:
@iwahdan88 

SYSTEM VERSION
==============
  FIRMWARE VERSION
    Bootloader0  : 5.1.1.0 [33080]
    Bootloader1* : 6.0.0.0-ER7 [35351]
    Bootloader2  : 6.0.0.0-ER7 [35351]
    NV Info      : 1.1,0,0
    Software     : 6.0.0.0-ER7 [35351] by robot-soft at 2018-03-08 13:38:13
    UE           : 6.0.0.0-ER7
  COMPONENTS
    ZSP0         : 1.0.0-251
    ZSP1         : 1.0.99-244

DNS seems to work now, but the whole thing is really unstable. The fipy can sometimes not even attach or connect to the mqtt broker.

iwahdan88, 2018-08-30:
@darox can you try upgrading to the new Firmware version (37781) following the steps [here](https://github.com/pycom/pycom-libraries/blob/master/lib/sqnsupgrade/README.md). and see if you still have problems?

darox, 2018-08-31:
@iwahdan88 I have three Fipys, none of them are working anymore, seems they can't find the cells anymore (NB-IoT and CAT M1).  I have tested in our live and lab network, same result.

iwahdan88, 2018-08-31:
@darox Is that after upgrading to new Firmware for LTE modem?

darox, 2018-08-31:
@iwahdan88 Yes

petervanderbeken, 2018-09-03:
DNS doesn't work for me either, on a NB-IoT connection with Orange Belgium. I'm on the latest public firmware:
```
SYSTEM VERSION
==============
  FIRMWARE VERSION
    Bootloader0  : 5.1.1.0 [33080]
    Bootloader1  : 5.1.1.0 [38638]
    Bootloader2* : 5.1.1.0 [38638]
    NV Info      : 1.1,0,0
    Software     : 6.0.0.0 [37781] by robot-soft at 2018-07-05 20:21:02
    UE           : 6.0.0.0
  COMPONENTS
    ZSP0         : 1.0.99-319
    ZSP1         : 1.0.99-302
```

```AT+CGCONTRDP``` shows empty strings starting from the 5th field (which is <DNS_prim_addr>).

https://forum.pycom.io/topic/2934/lte-nb-iot-dns/5 notes that adding ePCO support to a network fixed it for someone, but I don't know how to figure out if that's what's missing in my connection.

dmartauz, 2018-09-23:
@iwahdan88
Is there already any solution for this issue?

jbrichau, 2018-12-09:
I'm experiencing this issue as well, using latest NB1-40343 firmware on Belgium's Orange NB-IOT. Would be great if this can be addressed. 
Output of `sqnsupgrade.info()`:
```
<<< Welcome to the SQN3330 firmware updater [1.2.0] >>>
>>> GPy with firmware version 1.18.1.r4
Your modem is in application mode. Here is the current version:
UE6.0.0.0
LR6.0.0.0-40343
```

fusspawn, 2018-12-17:
Thanks for the reports. 

The firmware team is aware of the issues around DNS querys and missing dns server entrys and are working to resolve them.

rohlan, 2019-08-03:
hi. we had similar issues with a 1nce card in germany and could workaround it:

just by setting nameservers as documented on https://docs.pycom.io/firmwareapi/micropython/usocket/#socket-dnsserver-dnsindex-ip-addr

this was tested on lte modem nb1 firmware 41019
with fipy firmware 1.20.0.rc11

Pupix, 2021-09-17:
I am having the same problem right now. If I keep the fipy connected to the PC and upload code changes, after a few uploads the DNS stops working. The result is that the device gets stuck waiting for the ntp to be synched.

I tried manually setting the nameservers, both with Google's and Cloudflare's, but nothing worked. The only fix I found so far is to hard reset the device by removing the cable and putting it back in.

closed 2023-09-12 by darox