pycom/pycom-libraries #58: MQTT set_last_will

FranciscoGJ, 2018-05-30:
hi!, how can i use wills in mqtt. Can you give me an example?

chrisvoo, 2018-05-30:
Hi Francisco,
I don't have an example, but probably you should just call set_last_will passing a topic and a message as parameters in that order, [like shown in the signature of the method](https://github.com/micropython/micropython-lib/blob/master/umqtt.simple/umqtt/simple.py#L47).
This should be called before `connect` method.
See also [this thread in the micropython forum](https://forum.micropython.org/viewtopic.php?t=3584).

FranciscoGJ, 2018-05-30:
wow! thanks, I had not set the  keepalive params, that was my problem.

sdaniel55, 2018-07-05:
Hi @FranciscoGJ, I think this issue has been resolved, so I'll close this. If you think otherwise, feel free to reopen it and add more details. Thanks!

closed 2018-07-05 by sdaniel55