pycom/pycom-libraries #59: SQN3330 firmware updater OSError: AT+SMSWBOOT=3,0 failed!

simonpfeifhofer, 2018-06-05:
Got the following error updating SQN3330 firmware:

```
<<< Welcome to the SQN3330 firmware updater >>>
Entering recovery mode
Traceback (most recent call last):
  File "main.py", line 3, in <module>
  File "/flash/lib/sqnsupgrade.py", line 126, in run
OSError: AT+SMSWBOOT=3,0 failed!
MicroPython v1.8.6-849-56d00234 on 2018-05-18; GPy with ESP32
```

mikolajo, 2018-06-28:
I have the same issue, have you found any solution?

tmcolby, 2018-07-01:
I also cannot seem to get the firmware to update.  I am using the uartmirror method.

>>> sqnsupgrade.run('/home/tyson/pycom/FIPY_GPY_CATM1_33988.dup', 921600, port='/dev/ttyUSB0')
<<< Welcome to the SQN3330 firmware updater >>>
Entering recovery mode
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/tyson/pycom/pycom-libraries/lib/sqnsupgrade/sqnsupgrade.py", line 126, in run
    raise OSError('AT+SMSWBOOT=3,0 failed!')
OSError: AT+SMSWBOOT=3,0 failed!

mikolajo, 2018-07-01:
so for me worked: putting the run segment into looped try, after couple iterations it got it right

sdaniel55, 2018-07-05:
Hi @simonpfeifhofer, did you manage to update the firmware? If not, please reopen this issue and I can help you with it.

closed 2018-07-05 by sdaniel55

sh4nnongoh, 2018-07-26:
Hi @sdaniel55 , I have problems with this detailed here.

https://forum.pycom.io/topic/3547/fipy-lte-nb-problems-sqn3330-firmware-updater

Is it really true that I need to update the LTE-M1 first before LTE-NB?

sdaniel55, 2018-07-27:
Hi @sh4nnongoh, I've replied to your post on the forum, so I'll just post it here too

> "Please check [this](https://forum.pycom.io/topic/3549/important-update-regarding-lte-modem-updates) post for updates about the modem upgrades. Hope this helps!"