pycom/pycom-libraries #60: where can i find the code source of the network.Bluetooth file?

MoustaphaDIENG, 2018-06-05:
I want to see the different methods of the class Bluetooth. which file I have to open? 
I use pymakr.
 Thanks!

Xykon, 2018-06-05:
You can run this code directly on the device:

```
from network import Bluetooth
help(Bluetooth)
```

Otherwise you'll have to look here:
https://github.com/pycom/pycom-micropython-sigfox/blob/master/esp32/mods/modbt.c

MoustaphaDIENG, 2018-06-05:
i only get a list of the fonctions with help(Bluetooth). I rather need to see the source code of each function of the Bluetooth class. 
ie: i work with the lopy4 board.

Xykon, 2018-06-05:
You have the [source code for the Bluetooth module](https://github.com/pycom/pycom-micropython-sigfox/blob/master/esp32/mods/modbt.c) I already linked earlier plus the [Bluetooth component of the IDF](https://github.com/pycom/pycom-esp-idf/tree/master/components/bt).

MoustaphaDIENG, 2018-06-06:
Thank you for the link. I looked at it and saw that the code is in c. But I thought that the libraries used on pymakr are in python. For example, I do not see the function Bluetooth.resolve_adv_data().
Is the link you gave me the one used by pymakr?

Xykon, 2018-06-06:
Bluetooth is a module that's integrated in the firmware, and the firmware is written in C (using an external SDK for the processor that is written in C/C++), including the function [Bluetooth.resolve_adv_data()](https://github.com/pycom/pycom-micropython-sigfox/blob/4481a1c283be97640592ea2a0440830e336064de/esp32/mods/modbt.c#L849)

closed 2018-06-19 by Xykon