pycom/pycom-libraries #64: Exception: Board not detected: I2C bus error (Pytrack only)

zredlined, 2018-08-14:
I'm not able to use the pytrack sensors (the board communicates fine through USB) with either of the fipy or gpy boards that I have. When I try to access a sensor on the Pytrack, whether it's the GPS or accelerometer, I get the error below. I've tried both stable and developer Pycom firmware updates, and am running the latest 0.0.8 DFU for the pytrack.

If I replace the pytrack with a pysense board and update the library imports, the pysense board is detected and the accelerometer code works fine. 

Any ideas what I'm missing? Is there something special I need to configure on the pytrack to work with Fipy or Gpy?

>>> Traceback (most recent call last):
  File "<stdin>", line 7, in <module>
  File "/flash/lib/pytrack.py", line 8, in __init__
  File "/flash/lib/pycoproc.py", line 93, in __init__
  File "/flash/lib/pycoproc.py", line 93, in __init__
Exception: Board not detected: I2C bus error

Boards tested:
Working: tested both Fipy and gpy + pysense (latest pysense_0.0.8.dfu and stable Pycom firmware)
Not working: tested both Fipy and gpy + pytrack (latest pytrack_0.0.8.dfu and stable or dev Pycom firmware)

Code tested:
```
import machine
from machine import I2C
from LIS2HH12 import LIS2HH12
from pytrack import Pytrack

py = Pytrack()
acc = LIS2HH12()

while True:
   print("Grabbing pitch")
   pitch = acc.pitch()
   roll = acc.roll()
   print('{},{}'.format(pitch, roll))
   time.sleep_ms(100)
```

sdaniel55, 2018-08-15:
Hi @zredlined 👋 

The only error I had was missing `import time`, but otherwise it worked fine. Could you confirm that you've inserted the board correctly into Pytrack? The LED and USB port should be on the same side. You can find an image with the correct setup [here](https://docs.pycom.io/chapter/gettingstarted/connection/gpy.html#pic).

zredlined, 2018-08-15:
Confirmed- I tried re-seating the board a few times following the image above using both Fipy and Gpy modules, neither worked. They work fine on my pysense, but I don't have a second pytrack.

catalinio, 2018-08-16:
hi @zredlined, what's your firmware version? `uos.uname()`
You don't happen to have some other type, Lopy/Wipy/Sipy modules, to try with the Pytrack?
Did Pytrack ever worked fine?

sdaniel55, 2018-08-21:
@zredlined I tried to reproduce it with your config, but it still worked fine. It's possible that your Pytrack has an  issue. Could you contact us at sales@pycom.io and someone can discuss the details with you. Thanks!