pycom/pycom-libraries #71: Wrong documentation

darox, 2018-08-30:
In the documentation to upgrade the firmware, there's a step called:
os.mkfs(sd)

this formats the sd card and therefore not true. 

After not executing this command, it worked!

sdaniel55, 2018-08-30:
Hi @darox 👋 

On which page did you find this issue?

darox, 2018-08-30:
@sdaniel55 https://github.com/pycom/pycom-libraries/blob/master/lib/sqnsupgrade/README.md

sdaniel55, 2018-08-30:
@darox First step: "Format your SD card as with the FAT file system and then copy the files onto the card using your computer" then mount it. Once you've set up the SD card, you transfer the upgrade files. If you follow the steps in their order, it works fine.

sdaniel55, 2018-08-30:
@darox I've changed the layout and added comments to the code example to avoid confusion