pycom/pycom-libraries #72: Sqnsupgrade via UART fails on FiPy: object 'LTE' has no attribute 'modem_upgrade_mode'

pirate, 2018-09-02:
Following these instructions: https://docs.pycom.io/tutorials/lte/firmware
(fyi there's also a broken link in those docs: https://github.com/pycom/pycom-libraries/lib/sqnsupgrade​ should be /tree/...)

```python
Uploading project (main folder)...
Not safe booting, disabled in settings
Reading file status
[1/5] Writing file lib/sqnsbr.py
[2/5] Writing file lib/sqnscodec.py
[3/5] Writing file lib/sqnscrc.py
[4/5] Writing file lib/sqnstp.py
[5/5] Writing file lib/test_sd.py
Upload done, resetting board...
OKets Jun  8 2016 00:22:57

rst:0x7 (TG0WDT_SYS_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff8028,len:8
load:0x3fff8030,len:1728
load:0x4009fa00,len:0
load:0x4009fa00,len:14584
entry 0x400a059c
Pybytes library: v0.9.0
Preparing modem for upgrade...
Attempting AT wakeup...
Starting STP (DO NOT DISCONNECT POWER!!!)
Session opened: version 1, max transfer 8192 bytes
Sending 54854 bytes: [########################################] 100%
Bootrom updated successfully, switching to upgrade mode
Attempting AT auto-negotiation...
Going into MIRROR mode... please close this terminal to resume the upgrade via UART
Traceback (most recent call last):
  File "main.py", line 2, in <module>
  File "/flash/lib/sqnsupgrade.py", line 575, in uart
  File "/flash/lib/sqnsupgrade.py", line 503, in upgrade_uart
  File "/flash/lib/sqnsupgrade.py", line 309, in __run
  File "/flash/lib/sqnsupgrade.py", line 461, in uart_mirror
AttributeError: type object 'LTE' has no attribute 'modem_upgrade_mode'
Pycom MicroPython 1.18.0.r1 [v1.8.6-849-9569a73] on 2018-07-20; FiPy with ESP32
Type "help()" for more information.
```

sdaniel55, 2018-09-03:
Hi @pirate 👋 

Thanks for reporting the link, I've fixed it. Have you tried both (SD card and UART) methods? Just to confirm, your modem is not bricked, right?

pirate, 2018-09-05:
Thanks! I don't think it's bricked, haven't tried the SD card method as I
don't have a spare SD card to do it with. I'll give uart another shot
though and report back.

GuilRies, 2018-09-13:
Same problem here.
Any luck?

sdaniel55, 2018-09-19:
Hi @pirate and @GuilRies 👋 

Could you make sure that you use the `stable` firmware? The latest version is `1.18.1.r1`. Let me know if that fixes your issue

firew0rks, 2018-10-11:
Using the latest firmware, as per @sdaniel55 suggestion worked. I upgraded my device using the Pycom Firmware Tool and upgraded the modem firmware using the UART method. I found that it only worked using the recovery method and referencing the `updater.elf` file on my computer instead of flashing it onto the device.

pirate, 2018-10-12:
I wont have a chance to check this for another month or so (I'm traveling now), you're welcome to close it and I can reopen if I'm still having issues.