pycom/pycom-libraries #75: Sqnsupgrade does not work

xbolshe, 2018-09-21:
```
>>> os.uname()
(sysname='FiPy', nodename='FiPy', release='1.18.1.r1', version='v1.8.6-849-b0520f1 on 2018-08-29', machine='FiPy with ESP32', lorawan='1.0.2', sigfox='1.0.1')
>>> from network import LTE
>>> lte = LTE()
>>> lte.send_at_cmd('ATI')
'\r\nSEQUANS Communications\r\nVZM20Q\r\nUE6.0.0.0\r\n\r\nOK\r\n'
>>> from machine import SD
>>> sd = SD()
>>> os.mount(sd, '/sd')
>>> os.listdir('/sd')
['NB1-38729.dup', 'updater.elf']
>>> import sqnsupgrade
>>> sqnsupgrade.run('/sd/NB1-38729.dup', '/sd/updater.elf')
<<< Welcome to the SQN3330 firmware updater >>>
Attempting AT wakeup...
Starting STP (DO NOT DISCONNECT POWER!!!)
Session opened: version 1, max transfer 8192 bytes
Sending 54854 bytes: [########################################] 100%
Bootrom updated successfully, switching to upgrade mode
Attempting AT auto-negotiation...
ERROR sending AT command... no response? b''
AT auto-negotiation failed! Exiting.
Unable to load updater from /sd/updater.elf
>>>
```

Xykon, 2018-09-22:
If you please power cycle the module and try again, do you get the same output?

xbolshe, 2018-09-22:
Yes, there is the same output after a power cycle.

Xykon, 2018-09-22:
OK thanks for checking... can you please try to run the command without specifying the updater.elf file (again after a power cycle?)

import sqnsupgrade
sqnsupgrade.run('/sd/NB1-38729.dup', debug=True)

xbolshe, 2018-09-22:
There is no response more than 5 minutes at the end of log shown below:

```
>>> from machine import SD
>>> sd = SD()
>>> os.mount(sd, '/sd')
>>> os.listdir('/sd')
['NB1-38729.dup', 'updater.elf']
>>> import sqnsupgrade
>>> sqnsupgrade.version()

SYSTEM VERSION
==============
  FIRMWARE VERSION
    Bootloader0  : 5.1.1.0 [33080]
    Bootloader1  : 5.1.1.0 [36417]
    Bootloader2* : 6.0.0.0 [37781]
    NV Info      : 1.1,0,0
    Software     : 6.0.0.0 [37781] by robot-soft at 2018-07-05 20:21:02
    UE           : 6.0.0.0
  COMPONENTS
    ZSP0         : 1.0.99-319
    ZSP1         : 1.0.99-302

>>> sqnsupgrade.run('/sd/NB1-38729.dup', debug=True)
File /sd/NB1-38729.dup has size 5438102
Modem state: 2
Resume: False Retry: False
<<< Welcome to the SQN3330 firmware updater >>>
Success1? True
Success2? True
mirror? False  recover? False  resume? False  direct? False  atneg_only? False bootrom? False
Initial prepartion complete...
Loading /sd/NB1-38729.dup
Attempting AT wakeup...
b'\r\nOK\r\n'
Entering recovery mode
AT+SMOD? returned 2
..
Starting STP ON_THE_FLY
Starting STP code upload
running m.wipe
running m.reset
magic=0x74736166, op=0x80, sid=0x0, plen=0x0, tid=0x0, hcrc=0x0, pcrc=0x0
Correct SRSP signature: 0x74736166
running m.open_session
Session opened: version 1, max transfer 8192 bytes
running push(m)
Sending 5438102 bytes: [                                        ]   1%
```

xbolshe, 2018-09-22:
And next:
```
Correct SRSP signature: 0x74736166
running m.open_session
Session opened: version 1, max transfer 8192 bytes
running push(m)
Sending 5438102 bytes: [                                        ]   1%SRSP header too small: 0
Code download failed, aborting!
```

Xykon, 2018-09-22:
OK, just to make sure do you have any code running on the module (anything in boot.py or main.py?)

After a power cycle, does the sqnsupgrade.version command still work?

I can see that the modem has been updated previously. Did you use the same microSD card for this? Do you have another card you can try?

xbolshe, 2018-09-22:
There are no codes in boot.py and main.py
Yes, sqnsupgrade.version works after a power circle.
Yes, I used the same microSD card.

Xykon, 2018-09-22:
OK thanks again for checking... your modem doesn't seem to behave as expected because I tried the same upgrade on another FiPy using the same firmware versions and I'm seeing different results. I'll have to do some checks and probably also involve Sequans for additional troubleshooting.

I'll let you know as soon as I can provide more details.

xbolshe, 2018-09-22:
Thanks a lot.

Xykon, 2018-09-22:
Could you please try the following to help me send an issue report to Sequans:

Power cycle the module and enter the following commands:

import time
from network import LTE
lte = LTE()
time.sleep(1)
lte.send_at_cmd('AT+SMOD?')
time.sleep(1)
lte.send_at_cmd('AT+SMSWBOOT=3,1')
time.sleep(5)
lte.send_at_cmd('AT+SMOD?')

Please post the full output here. Thanks a lot.

xbolshe, 2018-09-22:
```
>>> import time
>>> from network import LTE
>>> lte = LTE()
>>> time.sleep(1)
>>> lte.send_at_cmd('AT+SMOD?')
'\r\n2\r\nOK\r\n'
>>> time.sleep(1)
>>> lte.send_at_cmd('AT+SMSWBOOT=3,1')
'\r\nERROR\r\n'
>>> time.sleep(5)
>>> lte.send_at_cmd('AT+SMOD?')
'\r\n2\r\nOK\r\n'
>>>
```

Xykon, 2018-09-22:
Thanks a lot... I have opened a case with Sequans as it's very unusual for 'AT+SMSWBOOT=3,1' to return ERROR.

sdaniel55, 2018-09-25:
Hi @xbolshe 👋 

Could you send us an email to support@pycom.io? This is probably a hardware issue, we can discuss the details via email. Could you put `[GitHub pycom-libraries #75]` in the subject line as well? I'll keep an eye out for your message 👀  Thanks!

closed 2018-09-25 by sdaniel55

pirate, 2018-11-10:
Fyi @sdaniel55 many other people are having this same issue: https://github.com/pycom/pycom-libraries/issues/81

sdaniel55, 2018-11-11:
@pirate Yes, we're looking into it

crushedy, 2020-04-13:
This also happened to me. I fixed it by restoring the modem to its factory setting:
from network import LTE
lte = LTE()
lte.factory_reset()