pycom/pycom-libraries #81: Sqnsupgrade recovery fails in FiPy

antonisgotsis, 2018-10-15:
When trying to upgrade the LTE NB-IOT firmware using the recovery mode (fw version: 1.18.r1, lte fw version: NB1-37781) I am getting the following:
```
>>> import sqnsupgrade
>>> sqnsupgrade.run('/sd/NB1-37781.dup', '/sd/updater.elf', debug=True)
File /sd/updater.elf has size 306076
File /sd/NB1-37781.dup has size 5258127
Modem state: None
Resume: False Retry: False
<<< Welcome to the SQN3330 firmware updater >>>
mirror? False  recover? False  resume? False  direct? False  atneg_only? False bootrom? True 
Initial prepartion complete...
Loading built-in recovery bootrom
Attempting AT wakeup...
b''
count=1
b''
count=2
b''
count=3
b''
count=4
b''
count=5
b''
count=6
b''
count=7
b''
count=8
b''
count=9
b''
count=10
b''
Success1? False
Unable to upgrade bootrom.
Success2? False
Unable to load updater from /sd/updater.elf
```
I suspect this is also related to Issue #75 (hardware issue).
How should I proceed?

humaxnerd, 2018-10-17:
Same issue here. My update stops at 1%.

Attempting AT wakeup...
Upgrader loaded successfully, modem is in upgrade mode
Attempting AT wakeup...
Starting STP ON_THE_FLY
Session opened: version 1, max transfer 8192 bytes

Sending 5996938 bytes: [                                        ]   0%
Sending 5996938 bytes: [                                        ]   0%
Sending 5996938 bytes: [                                        ]   0%
Sending 5996938 bytes: [                                        ]   0%
Sending 5996938 bytes: [                                        ]   0%
Sending 5996938 bytes: [                                        ]   0%
Sending 5996938 bytes: [                                        ]   0%
Sending 5996938 bytes: [                                        ]   0%
Sending 5996938 bytes: [                                        ]   0%
Sending 5996938 bytes: [                                        ]   0%
Sending 5996938 bytes: [                                        ]   0%
Sending 5996938 bytes: [                                        ]   0%
Sending 5996938 bytes: [                                        ]   0%
Sending 5996938 bytes: [                                        ]   0%
Sending 5996938 bytes: [                                        ]   1%
Sending 5996938 bytes: [                                        ]   1%
Sending 5996938 bytes: [                                        ]   1%
Sending 5996938 bytes: [                                        ]   1%
Sending 5996938 bytes: [                                        ]   1%
Sending 5996938 bytes: [                                        ]   1%
Sending 5996938 bytes: [                                        ]   1%
Sending 5996938 bytes: [                                        ]   1%
Sending 5996938 bytes: [                                        ]   1%
Sending 5996938 bytes: [                                        ]   1%
Sending 5996938 bytes: [                                        ]   1%
Sending 5996938 bytes: [                                        ]   1%
Sending 5996938 bytes: [                                        ]   1%SRSP header too small: 0
Code download failed, aborting!

Any ideas?

martijns, 2018-10-22:
Same issue here. Upgraded once before to CATM1 firmware. Also the ERROR on AT+SMSWBOOT=3,1 as mentioned in #75.

Version info:

```
>>> sqnsupgrade.version()

SYSTEM VERSION
==============
  FIRMWARE VERSION
    Bootloader0  : 5.1.1.0 [33080]
    Bootloader1* : BUILD5.1.1.0-33988 by robot-soft at 2017-12-11 15:06:33
    Bootloader2  : BUILD5.1.1.0-33988 by robot-soft at 2017-12-11 15:06:33
    NV Info      : 1.1,0,0
    Software     : 5.1.1.0 [33988] by robot-soft at 2017-12-11 15:04:15
    UE           : Unknown
  COMPONENTS
    ZSP0         : 1.0.99-13028
    ZSP1         : 1.0.98-12295

>>> os.uname()
(sysname='FiPy', nodename='FiPy', release='1.18.1.r1', version='v1.8.6-849-b0520f1 on 2018-08-29', machine='FiPy with ESP32', lorawan='1.0.2', sigfox='1.0.1')
```

ka5ko, 2018-10-28:
I have same issue with Gpy .   Does anyone has any update ?  Thank you

>>> sqnsupgrade.run('/sd/CATM1-38638.dup','/sd/updater.elf', debug=True)
File /sd/updater.elf has size 306076
File /sd/CATM1-38638.dup has size 5996938
Modem state: 2
Resume: False Retry: False
<<< Welcome to the SQN3330 firmware updater >>>
mirror? False  recover? False  resume? False  direct? False  atneg_only? False bootrom? True
Initial prepartion complete...
Loading built-in recovery bootrom
Attempting AT wakeup...
b'\r\nOK\r\n'
Entering recovery mode
AT+SMOD? returned 2
Starting STP (DO NOT DISCONNECT POWER!!!)
Starting STP code upload
running m.wipe
running m.reset
magic=0x74736166, op=0x80, sid=0x0, plen=0x0, tid=0x0, hcrc=0x0, pcrc=0x0
Correct SRSP signature: 0x74736166
running m.open_session
Session opened: version 1, max transfer 8192 bytes
running push(m)
Sending 54854 bytes: [########################################] 100%
running dev.set_timeout(2)
running m.reset(True)
Bootrom updated successfully, switching to upgrade mode
Success1? True
mirror? False  recover? True  resume? True  direct? False  atneg_only? False bootrom? False
Initial prepartion complete...
Loading /sd/updater.elf
Entering recovery mode
AT+SMOD? returned
Attempting AT auto-negotiation...
b''
count=1
b''
count=2
b''
count=3
b''
count=4
b''
count=5
b''
count=6
b''
count=7
b''
count=8
b''
count=9
b''
count=10
b''
ERROR sending AT command... no response? b''
AT auto-negotiation failed! Exiting.
Success2? False
Unable to load updater from /sd/updater.elf
>>> sqnsupgrade.run('/sd/CATM1-38638.dup', debug=True)
File /sd/CATM1-38638.dup has size 5996938
Modem state: 2
Resume: False Retry: False
<<< Welcome to the SQN3330 firmware updater >>>
Success1? True
Success2? True
mirror? False  recover? False  resume? False  direct? False  atneg_only? False bootrom? False
Initial prepartion complete...
Loading /sd/CATM1-38638.dup
Attempting AT wakeup...
b'\r\nOK\r\n'
Entering recovery mode
AT+SMOD? returned 2
..
Starting STP ON_THE_FLY
Starting STP code upload
running m.wipe
running m.reset
magic=0x74736166, op=0x80, sid=0x0, plen=0x0, tid=0x0, hcrc=0x0, pcrc=0x0
Correct SRSP signature: 0x74736166
running m.open_session
Session opened: version 1, max transfer 8192 bytes
running push(m)
Sending 5996938 bytes: [                                        ]   1%

>>> os.uname()
(sysname='GPy', nodename='GPy', release='1.18.1.r1', version='v1.8.6-849-b0520f1 on 2018-08-29', machine='GPy with ESP32')

pirate, 2018-11-10:
Same issue here with the FiPy:

```python
>>> sqnsupgrade.run('/sd/CATM1-39529.dup', '/sd/updater.elf')
<<< Welcome to the SQN3330 firmware updater >>>
Attempting AT auto-negotiation...
Session opened: version 1, max transfer 2048 bytes
Sending 429340 bytes: [########################################] 100%
Attempting AT wakeup...
Upgrader loaded successfully, modem is in upgrade mode
Attempting AT wakeup...
Starting STP ON_THE_FLY
Session opened: version 1, max transfer 8192 bytes
Sending 6059510 bytes: [                                        ]   1%SRSP header too small: 0
Code download failed, aborting!
```