pycom/pycom-libraries #83: Lopy4 TTN Join fail - Code examples NOT working

Synapsecontrols, 2018-10-23:
Hi,

We need an example code that works / with Lopy4 - The things Network for US frequency, we have several projects running with Pycom boards, there are too many flaws with these devices.

This code works with Lopy but NOT with Lopy4

from network import LoRa
import socket
import binascii
import time
import machine
import pycom
import os

pycom.heartbeat(False)

lora = LoRa(mode=LoRa.LORAWAN, public=True, adr=False)

for index in range(0, 8): #turn off channels 0-7
lora.remove_channel(index)

for index in range(16, 65): #turn off channels 16-64
lora.remove_channel(index)

for index in range(66, 72): #turn off channels 66-71
lora.remove_channel(index)

auth = (bytes([ 0x70, 0xB3, 0xD5, 0x7E, 0xD0, 0x00, 0xB5, 0x78 ]),
bytes([ 0xCE, 0xF9, 0x6A, 0x8F, 0x9D, 0x28, 0x1D, 0x51, 0x4C, 0xEC, 0xF1, 0x14, 0xE3, 0xA9, 0xD3, 0x4C ]))

print("Starting Join...")
lora.join(activation=LoRa.OTAA, auth=auth, timeout=0)
x=0

while (not lora.has_joined() and x<10):
time.sleep(2.5)
print('Lora Joining...');
x=x+1
print('Lora has joined?= ', lora.has_joined())

s = socket.socket(socket.AF_LORA, socket.SOCK_RAW)
s.setsockopt(socket.SOL_LORA, socket.SO_DR, 3)
lora.remove_channel(65)
s.setblocking(False)
time.sleep(5.0)

if (lora.has_joined()):
print('Joined...')

while True:
s.send(bytes([0x01]))
time.sleep(60)

I appreciate your help.

(sysname='LoPy', nodename='LoPy', release='1.18.1.r1', version='v1.8.6-849-b0520f1 on 2018-08-29', machine='LoPy with ESP32', lorawan='1.0.2')

catalinio, 2018-11-09:
Hi @Synapsecontrols, it should be no difference between Lopy and Lopy4.
Are you sure, you're having the Lora antenna in the 868/915Mhz, and not in 433Mhz connector?
In Lopy4 the same connector used in Lopy1 for 868Mhz is 433Mhz(the one close to Reset button).
https://docs.pycom.io/datasheets/development/lopy4
https://blobscdn.gitbook.com/v0/b/gitbook-28427.appspot.com/o/assets%2F-LIL0iGdl11z7Jos_jpX%2F-LKN7scqKKkb6TqB3uYO%2F-LKN85nYahseE5Es3d4z%2Flopy4-pinout.png?generation=1534782115650412&alt=media

I'm sorry for the long response time.

fusspawn, 2018-12-17:
Hi @Synapsecontrols Is this still an issue your seeing?

Synapsecontrols, 2018-12-17:
Hi,

I had to buy another lopy4 to check that it was not damaged. Effectively,
it was damaged from factory. I already managed to connect to the server.
Thank you very much for the support.

Regards.