pycom/pycom-libraries #87: MQTT error ECONNRESET

robmarkcole, 2018-11-07:
Hi I am receiving the following error with source code [here](https://github.com/robmarkcole/pycom-projects/blob/master/PMS5003%20air%20quality%20sensor/main.py):
```
>>> client.publish("wipy/", "test")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/flash/lib/mqtt.py", line 110, in publish
OSError: [Errno 104] ECONNRESET
```
This error is [reported](https://forum.pycom.io/topic/3648/umqtt-robust-stalls-out-on-fipy) in a couple of different threads on the forums, and [perhaps can be resolved](https://forum.pycom.io/topic/2311/mqtt-oserror-issue/3) by implementing umqtt-robust.

sdaniel55, 2018-11-07:
Hi @robmarkcole 👋 Can I close this ticket then?

robmarkcole, 2018-11-07:
HI @sdaniel55 if the error cannot be handled to provide more debugging info, then yes please close. However I could have saved 1 hr with some more debugging info and I'm sure i won't be the last to get this  error. Therefore if it could be handeld better perhaps this can be a feature request?
Cheers
Robin

sdaniel55, 2018-11-07:
@robmarkcole I'll mark it as a feature request then 😉 Do you have anything else to add? There's a [feature request template](https://github.com/pycom/pycom-libraries/issues/new?template=feature_request.md) for issues, if you need some guidance.

robmarkcole, 2018-11-07:
To define the feature request: more informative messages on errors, thanks :-)

closed 2018-11-07 by robmarkcole