pycom/pycom-libraries #89: MQTT subscribe

Shedaim, 2018-12-03:
## What are the steps to reproduce this issue?
Subscribing to an MQTT broker using: 
1. mqtt_object.set_callback(func)
2. mqtt_object.subscribe(topic)

## What happens?
Code waits in a while 1 untill op == 0x90.
I get a response from the server which matches op == 0x92 (9 = Subscribe ACK, 2= reserved).
The server accepts the subscription, but Pycom gets stuck in an infinite loop.
Same happens if no answer arrives to the subscription request. 

## What were you expecting to happen?
Subscription ACK arrives and the code keeps running.

## What versions of software are you using?
sysname='GPy', nodename='GPy', release='1.18.1.r4', version='v1.8.6-849-dfa1176 on 2018-11-12', machine='GPy with ESP32

fusspawn, 2018-12-17:
@iwahdan88 Is this one for you? not sure who handles the mqtt libs internally.

Shedaim, 2019-01-22:
Have you seen the issue?