pycom/pycom-libraries #97: Add lux calculations to the LTR-329ALS sensor class

amcewen, 2019-04-06:
The light sensor on the pySense boards reports two light values from the LTR-329ALS sensor.

Reading the [datasheet for the sensor](https://optoelectronics.liteon.com/upload/download/DS86-2014-0006/LTR-329ALS-01_DS_V1.pdf) that's because there are two sensors sensitive to different wavelengths of light.

The datasheet mentions that there is a formula for calculating the lux level from those two readings.  It would be good to have the lux reading available from the `LTR329ALS01` class as well as the raw light readings.

amcewen, 2019-04-06:
The datasheet doesn't actually include Appendix A, but i received a copy of it from LiteON when I asked them about it.

I've implemented the new method in my fork of the libraries, in https://github.com/mcqn/pycom-libraries/commit/3faed579996044865ab9ea84625852782d5cad97

Happy to submit a pull request, but at the moment I've already got PR #95 outstanding, so it looks like that would need to be accepted/rejected first?

Kriskras99, 2019-12-01:
Your implementation does not work if light_level == (0, 0), because the ratio calculation would be a division by zero (ZeroDivisionError)

amcewen, 2019-12-02:
Yes, sorry I forgot to update this issue - that bug was fixed in the next commit to my fork - https://github.com/mcqn/pycom-libraries/commit/8d655f9990f1a6c642ba8844d18ccb2b20556cf8 - a few weeks later.

MoraO, 2020-04-01:
Hello Adrian, could you share the file with the formula please?, I am working on a Light sensor using the LTR-329ALS-01 on my job and I need a reliable reference for my programming algorithm. The work will be share on github in C language and libraries for PIC mirochip microcontrollers. I really appreciate your help.

JobStokkermans, 2023-06-28:
Hello @amcewen Could you share appendix A with me? I am working on the LTR-329ALS-01 as well.