pycom/pycom-libraries #98: Pybytes constantly disconnects, requires manual restart

Craigzyc, 2019-04-22:
## What are the steps to reproduce this issue?
1. Add device to pybytes FW1.18.2R3
2. Modify main.py in the flash and push changes
3. be disappointed it doesn't come back online

## What happens?
Never comes back online. Needs a manual restart every time

## What were you expecting to happen?
Come back online

## Any logs, error output, etc?
*(If it’s long, please paste to https://gist.github.com and insert the link here)*
No

## Any other comments?
The documents are unclear but state that after 1.16 the pybytes library is included in the /frozen modules. Does that mean I can use the more recent development builds if I setup pybytes manually?
Pybytes is great in theory and would actually be quite useful if it supported firmware updates over cellular and worked as expected.

## What versions of software are you using?
Fipy
FW 1.18.2.R3

jirikrepl, 2019-04-23:
Hi, @Craigzyc thank you for reporting this issue.

Are you connected with FiPy over Wi-Fi and then using the flash tab?
I tried and it seems to be working fine for me.

At this moment we haven't made "pybytes-library" (don't confuse with pybytes-libraries - pysense/pytract lib.) repository public.

I mean the pybytes-library code is publicly available here in the /frozen folder of the firmware's pybytes-master branch https://github.com/pycom/pycom-micropython-sigfox/tree/pybytes-master/esp32/frozen/Base

The other way how to get pybytes-library at this moment is to go to device->Settings->download pybytes library. But that will still download just a source code of the master branch of the private pybytes-library repository. Furthermore, python files won't start with `_` and won't be just in the root folder.

Currently, we don't have pybytes-library python development code public.

We are aware of current Pybytes limitations and it's potential. We're working hard to fix all the issues

jirikrepl, 2019-04-23:
If you are still facing the problem with the connection, maybe we can set up a call.

macklin-fluehr, 2019-10-02:
Do we have an update on this issue? I have the same problem.

jirikrepl, 2019-10-02:
Are you using the latest firmware? Are you connected to your device over WiFi?

macklin-fluehr, 2019-10-02:
Yes. Using firmware version 1.20.0.R3, which looks up to date. Connected over Wifi. Have about 15 or so seconds of connection before disconnecting.

jirikrepl, 2019-10-03:
But we haven't publicly released firmware 1.20.x yet. Where and how you got this firmware, please?
(though we will release 1.20.x in a few days)

Up to this date, you can get Pybytes firmware 1.18.2.r6 with the firmware updater.

Please follow the steps described in these two pages (steps from the first page continue on the second page):
https://docs.pycom.io/pybytes/connect/
https://docs.pycom.io/pybytes/connect/quick/

[image]

robert-hh, 2019-10-03:
All firmware versions are made available by Pycom. For FiPy e.g. here: https://software.pycom.io/downloads/FiPy.html
Using the firmware updater, you can also install release candidates, like V1.20.0.rc13 etc.