pycom/pycom-libraries #100: Lora Mesh example is crashing

ricardolimaui, 2019-05-14:
## What are the steps to reproduce this issue?
1. run the two examples provided in two different windows
2. main.py
3. main_border_router.py

## What happens?
it successfully start but (the main.py), After a while it gives the following error

```
42: State detached, single true
Neighbors found: None
opened: 0x3f9512a0
Traceback (most recent call last):
  File "main.py", line 58, in <module>
TypeError: function takes 3 positional arguments but 2 were given
```

## What were you expecting to happen?
waiting forever for the mesh maybe

## Any logs, error output, etc?
```
LoRa MAC: Mac_from_main_node
1: State detached, single true
3: State detached, single true
6: State detached, single true
8: State detached, single true
11: State detached, single true
14: State detached, single true
16: State detached, single true
19: State detached, single true
21: State detached, single true
24: State detached, single true
27: State detached, single true
29: State detached, single true
32: State detached, single true
34: State detached, single true
37: State detached, single true
40: State detached, single true
42: State detached, single true
Neighbors found: None
opened: 0x3f9512a0
Traceback (most recent call last):
  File "main.py", line 58, in <module>
TypeError: function takes 3 positional arguments but 2 were given
Pycom MicroPython 1.20.0.rc10 [v1.9.4-350eac8] on 2019-04-16; LoPy4 with ESP32
Type "help()" for more information.
```

```
>>> LoRa MAC: Mac_from_border_router
31: State detached, single true
33: State detached, single true
36: State detached, single true
38: State detached, single true
41: State detached, single true
44: State detached, single true
46: State detached, single true
Neighbors found: None
IPs: ['fdde:ad00:beef:0:0:ff:fe00:a401', 'fdde:ad00:beef:0:8341:2090:c8e4:f279', 'fe80:0:0:0:72b3:d549:9777:124d']
opened: 0x3f951b20
Created socked for (::, 1235)
IPs: ['fdde:ad00:beef:0:0:ff:fe00:a401', 'fdde:ad00:beef:0:8341:2090:c8e4:f279', 'fe80:0:0:0:72b3:d549:9777:124d']
BRs: []
>
Pycom MicroPython 1.20.0.rc10 [v1.9.4-350eac8] on 2019-04-16; LoPy4 with ESP32
Type "help()" for more information.
```

## Any other comments?
I did not change anything in the code, just run the example. I only inserted the Loramesh.py inside of a folder called lib.
I also tried with the main from border router, but it seems like don't gave any error but also don't give any good result in terms of Neighbors

```
>>> LoRa MAC: Mac_from_border_router
31: State detached, single true
33: State detached, single true
36: State detached, single true
38: State detached, single true
41: State detached, single true
44: State detached, single true
46: State detached, single true
Neighbors found: None
IPs: ['fdde:ad00:beef:0:0:ff:fe00:a401', 'fdde:ad00:beef:0:8341:2090:c8e4:f279', 'fe80:0:0:0:72b3:d549:9777:124d']
opened: 0x3f951b20
Created socked for (::, 1235)
IPs: ['fdde:ad00:beef:0:0:ff:fe00:a401', 'fdde:ad00:beef:0:8341:2090:c8e4:f279', 'fe80:0:0:0:72b3:d549:9777:124d']
BRs: []
>
Pycom MicroPython 1.20.0.rc10 [v1.9.4-350eac8] on 2019-04-16; LoPy4 with ESP32
Type "help()" for more information.
```

## What versions of software are you using?
I am using a brand new lopy4 with a expansion board vrsion3.1 and I also have a pysense

Anyone can help me ? Don't know if it is really a bug

ricardolimaui, 2019-05-14:
the line were it gave the error contains this mesh.mesh.rx_cb(receive_pack)

L-Gibson, 2019-05-15:
Try using 1.20.0.rc7, fixed the issue for me.

ricardolimaui, 2019-05-15:
Hello again
yes L-Gibson 1.20.0.rc7 fixed the problem for sure. Thank you for your help
But know I am facing another problem

in the main node appear 
`2647: State leader, single true, IP fdde:ad00:beef:0:2592:49a4:52a9:d46a
1 neighbors, IPv6 list: ['fdde:ad00:beef:0:0:ff:fe00:6002']
Ping OK from neighbor fdde:ad00:beef:0:0:ff:fe00:6002
Sent message to fdde:ad00:beef:0:0:ff:fe00:6002` 
But then in the main_border_router appears an error

`Neighbors found: None
IPs: ['fdde:ad00:beef:0:0:ff:fe00:6002', 'fdde:ad00:beef:0:d369:349a:cd66:3399', 'fe80:0:0:0:72b3:d549:9777:124d']
Traceback (most recent call last):
  File "<stdin>", line 62, in <module>
TypeError: can't convert tuple to int
>
Pycom MicroPython 1.20.0.rc7 [v1.9.4-2833cf5] on 2019-02-08; LoPy4 with ESP32
Type "help()" for more information.
>>> 
>>> `

L-Gibson, 2019-05-15:
Seems like main only works on 1.20.0.rc7 and main_border_router works on the latest firmware. Though it seems that main_border_router contains all of the functionality in main (or it can be copied over)

ricardolimaui, 2019-06-19:
sorry about taking so much time. Yes indeed with that versions it works but the border router example code does not contemplate the connection to the internet right? For example, a node will be able to connect to the internet throught Loramesh reaching the border router ? 
Because seems like with the current code it does not find any border router

Thank you

ricardolimaui, 2019-08-01:
Any news about pymesh ? Thank you

catalinio, 2019-08-02:
Hi @ricardolimaui,

I've answered on the forum: https://forum.pycom.io/topic/4449/pymesh-updates/20

For the record, the initial problem (error on `mesh.rx_cb()` function) on release rc8 (I think) I've added another argument, as explained in docs: https://docs.pycom.io/firmwareapi/pycom/network/lora/pymesh/#mesh-rx-cb-handler-argument

Did I answer your question?

Cheers,
Catalin

ricardolimaui, 2019-08-20:
Hi @catalinio upgrading to the  1.20.0.rc12 this error comes again 

>  AttributeError: 'LoRa' object has no attribute 'Mesh'

Where we can find the pymesh version where the message are widespread in the network ?
The available one seems like it don't relay the message but only send the ack.

Which are the methology they use to chose of being a router leader or child?

I am trying to find more documentation, but not easy 

Could you provide it ? 

Thank you

catalinio, 2019-08-21:
Hi @ricardolimaui,

In the version `1.20.0.rc12` the Pymesh module(`LoRa.Mesh` class) was excluded from build, due to increased binary footprint. We've fixed this issue, but not yet released.
So, we'll release a separate `pymesh` firmware binary image (next to `stable`, `development` and `pybytes`); these days we're verifying the final image.

Regarding documentation, indeed I need to update it. Meanwhile I've put here some links on openthread core principles: https://docs.pycom.io/tutorials/lora/lora-mesh/; furthermore in the [OT cli](https://github.com/openthread/openthread/tree/master/src/cli) you have most of the commands available thru [pymesh.cli('command') function](https://docs.pycom.io/firmwareapi/pycom/network/lora/pymesh/#mesh-cli).

Cheers,
Catalin

ricardolimaui, 2019-08-21:
Thank you for quick support @catalinio 
Waiting for the new image :D
Just a note In the documentation there is at least a dead link.
"the complete list of commands is here." The link in here "https://github.com/openthread/openthread/blob/c482301ec73b80985445102e4d0a936346172ddb/src/cli/README" is dead(openthread changed the path ?).

ulrikjohansson, 2020-11-27:
Hi!
This seems to still (or again?) be a problem in newer firmware releases, atleast for me.
I'm trying to set up pybytes/pymesh manually in my own main.py instead of having it autostarted, so I've skipped updating my LoPy4 with the mesh specific firmware from pybytes.pycom.io. It seems like network.LoRa.Mesh() does not exist on the 1.20.2.r2 firmware I can find in the Windows Firmware updater.
I want to avoid the firmware from the pymesh OTA update, since it's insisting on using a release candidate version of the firmware (1.20.2.rc11). There does seem to exist an updated firmware on the server though, it's just not the version pybytes is pushing (see https://software.pycom.io/downloads/appimg/firmware_LoPy4_1.20.2.r2-pymesh.bin)

ricardolimaui, 2021-05-12:
Any news on this? As I can see there are more open issues similar to this one.

peter-pycom, 2021-05-12:
Could you retest with the pymesh firmware that's available via Pybytes? Lora mesh is only supported there