pycom/pycom-libraries #106: _udp_thread exception and the node gives has_joined() always false

skyluke86, 2019-10-11:
Hello,
I'm facing with two issues that to be honest i don't know if they are correlated.

First of all i have a LoPy as LoRa gateway connected to the TTN server via Wi-Fi on an expansion board 3.1
The release installed is:
Pycom MicroPython 1.20.1 [v1.11-12f4ce0] on 2019-10-06; LoPy4 with ESP32
Pybytes Version: 1.1.2

Second i have a FiPy as LoRa node with Pysense as board.
The release installed is:
Pycom MicroPython 1.20.1 [v1.11-12f4ce0] on 2019-10-06; FiPy with ESP32
Pybytes Version: 1.1.2

As i told from TTN, the gateway results connected but when i run the LoPy i got this error from console:

[     2.847] Starting LoRaWAN nano gateway with id: b'3C71BFFFFE8765D4'
[     4.264] WiFi connected to: Parappapa
[     4.272] Syncing time with pool.ntp.org ...
[     4.331] RTC NTP sync complete
[     4.373] Opening UDP socket to router.eu.thethings.network (52.169.76.203) port 1700...
[     4.393] Setting up the LoRa radio at 868.1 Mhz using SF7BW125
[     4.406] LoRaWAN nano gateway online
[     4.413] You may now press ENTER to enter the REPL
Unhandled exception in thread started by <bound_method>
Traceback (most recent call last):
  File "/flash/lib/nanogateway.py", line 451, in _udp_thread
  File "/flash/lib/nanogateway.py", line 450, in _udp_thread
AttributeError: 'OSError' object has no attribute 'errno'

Anyone has an idea how to solve?

Second problem is the FiPy LoRa node gives always false from has_joined().

I'm using the example code for both.

Any idea?

robert-hh, 2019-10-11:
I cannot locate the line mentioned in the error log, at least not in the file that is in pycom-libraries/examples/lorawan-nano-gateway. Which file are you using?

skyluke86, 2019-10-11:
Thanks for the answer, i'm talking about this file:
pycom-libraries/examples/lorawan-nano-gateway/nanogateway.py
I get the exception from the line 403:
data, src = self.sock.recvfrom(1024)

And the Unhandled exception from line 450, 451

skyluke86, 2019-10-11:
I forgot to say that the FiPy node from TTN results connected even if has_joined() gives always false.

robert-hh, 2019-10-11:
The problem I face is, that in the nanogateway.py version on the net line 450 is a comment.
https://github.com/pycom/pycom-libraries/blob/master/examples/lorawan-nano-gateway/nanogateway.py#L450
About the Fipy being connected: there may be another TTN gateway in reach which made the connection. In TTN, every node can connect to every gateway.

skyluke86, 2019-10-11:
I had a version with an offset of some line.
Anyway i restored the original file from repository and now the lines are:
Unhandled exception in thread started by <bound_method>
Traceback (most recent call last):
  File "/flash/lib/nanogateway.py", line 446, in _udp_thread
  File "/flash/lib/nanogateway.py", line 445, in _udp_thread
AttributeError: 'OSError' object has no attribute 'errno'

skyluke86, 2019-10-11:
> About the Fipy being connected: there may be another TTN gateway in reach which made the connection. In TTN, every node can connect to every gateway.

I switched off the gateway and let the FiPy node on, no refresh status from TTN.

robert-hh, 2019-10-11:
At least now the line numbers matched. No, that cannot work. ex has no such an attribute. Looks like this bug was in the code since ever. I do not know what the intention was, but you can safely delete line 445. That may only may cause more non-errors printed.

robert-hh, 2019-10-11:
Forget that with the deletion of the line. It's a bad idea. It seems that earlier version had the errno argument.

robert-hh, 2019-10-11:
Ok. In versions before 1.20.1 the exception object had an attribute errno. That's gone. Until that is fixed, you either have to go back to 1.20.0.rc13 or 1.18.2.r7 (links here https://forum.pycom.io/topic/2777/downgrading-firmware-advanced-users), or replace both lines 445 and 446 with a pass statement.

skyluke86, 2019-10-11:
Ok thanks.
Instead for the has_joined() issue, i swapped from OTAA to ABP and i can see the frame counter increase from TTN site. I did this test several times and with OTAA no works, with ABP works.

robert-hh, 2019-10-11:
I forgot to tell that. The nanogateway code is notoriously unreliable in delivering downlink packets, and it has very limited functionality. It cannot keep up with the timing precision required by LoRa.
OTAA requires a downlink packet, ABP does not. If you want to use LoRa with your own gateway, get one which supports the full service.

robert-hh, 2019-10-11:
OK. You can try to change line 445 to:
if ex.args[0] != errno.EAGAIN:

skyluke86, 2019-10-11:
Thanks for the help

closed 2019-10-11 by skyluke86

zetxx, 2020-01-20:
> OK. You can try to change line 445 to:

i can confirm that this works