pycom/pycom-libraries #108: Examples contain bugs

dbrosy, 2019-10-13:
(Hi! 👋  Thanks for reporting an issue! Please make sure you click the link above to view the issue guidelines, then fill out the blanks below.)

## What are the steps to reproduce this issue?
1. Download latest pycom-libraries
2. Install lorawan-nano-gateway example (nanogateway.py) to lopy4
3. Restart

## What happens?

'''
Unhandled exception in thread started by <bound_method>
Traceback (most recent call last):
  File "nanogateway.py", line 446, in _udp_thread
  File "nanogateway.py", line 445, in _udp_thread
AttributeError: 'OSError' object has no attribute 'errno'
'''

## What were you expecting to happen?
I was expecting the example to just work!!

## Any logs, error output, etc?
*(If it’s long, please paste to https://gist.github.com and insert the link here)*

## Any other comments?
I have edited line 446 from:
'''
                if ex.errno != errno.EAGAIN:
'''
to 
'''
                if ex.args[0] != errno.EAGAIN:
'''

this has resolved the issue, but should not OSError have an attribute - errno?

I have also noted a number of issues on docs.pycom.com where examples or references not working or containing formatting errors

https://docs.pycom.io/firmwareapi/micropython/usocket/ (line 22 of example)

## What versions of software are you using?
- **Board type and hardware version:**
- **`os.uname()` output:**
(sysname='LoPy4', nodename='LoPy4', release='1.20.1.r1', version='v1.11-3138a13 on 2019-10-08', machine='LoPy4 with ESP32', lorawan='1.0.2', sigfox='1.0.1', pybytes='1.1.3')

jaray200, 2019-11-19:
I can confirm that this has allowed me to establish a connection to my local chirpstack server, though at the time of writing this I have not been able to see the gateway within the chirpstack application.

Pablo849, 2019-11-28:
Same error here

robert-hh, 2019-11-28:
See this PR https://github.com/pycom/pycom-libraries/pull/107, which seems not to find any attention by Pycom.

doniks, 2020-02-06:
#107 is merged. Please retest