pycom/pycom-libraries #117: lorawan-nano-gateway crashes after some time

pirminwenk, 2020-04-17:
Hello guys

I have a LoPy4 v1.20.2.rc6 with Expansion board expansion31_0.0.11
Tried to load the example project "lorawan-nano-gateway"

After couple of hours (between 1 and 8) the LoPy crashes with following (decoded) stacktrace:
```
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

PC: 0x4010466c: mp_obj_str_binary_op at ../py/objstr.c line 1500
EXCVADDR: 0x00000001

Decoding stack results
0x4010466c: mp_obj_str_binary_op at ../py/objstr.c line 1500
0x400debf5: uECC_vli_modMult at /Users/ehlers/pycom/pycom-esp-idf/components/micro-ecc/micro-ecc/uECC.c line 601
0x400dec5f: uECC_vli_modMult at /Users/ehlers/pycom/pycom-esp-idf/components/micro-ecc/micro-ecc/uECC.c line 599
0x400df131: uECC_verify at /Users/ehlers/pycom/pycom-esp-idf/components/micro-ecc/micro-ecc/uECC.c line 1497
0x4010256d: mp_obj_int_unary_op at ../py/objint_mpz.c line 148
0x400fc45a: mp_compile at ../py/compile.c line 3385
0x400f8bfd: mp_lexer_to_next at ../py/lexer.c line 549
0x40104177: str_strip at ../py/objstr.c line 858
0x400fc4e4: mp_compile at ../py/compile.c line 3409
0x400f8bfd: mp_lexer_to_next at ../py/lexer.c line 549
0x400f8c2a: mp_lexer_to_next at ../py/lexer.c line 548
0x400df1da: uECC_verify at /Users/ehlers/pycom/pycom-esp-idf/components/micro-ecc/micro-ecc/uECC.c line 1514
0x400df44d: uECC_verify at /Users/ehlers/pycom/pycom-esp-idf/components/micro-ecc/micro-ecc/uECC.c line 1555
0x400de08b: image_load at /Users/ehlers/pycom/pycom-esp-idf/components/bootloader_support/src/esp_image_format.c line 423
```

Tried to use different WLANs.
I also tried to completely reset the LoPy4 like described here:
https://forum.pycom.io/topic/4307/guru-meditation-error-pycom-wipy/35

Does anyone have the same problem?
Any ideas what to try?

Appreciate your help.

pirminwenk, 2020-04-17:
Just had another stacktrace with the same software:
```
PC: 0x4010a38b: mp_execute_bytecode at ../py/vm.c line 1338
EXCVADDR: 0x00000001

Decoding stack results
0x4010a38b: mp_execute_bytecode at ../py/vm.c line 1338
0x400e0fc5: mp_hal_delay_ms at hal/esp32_mphal.c line 218
0x400e1036: mp_hal_stdin_rx_chr at hal/esp32_mphal.c line 148
0x40108211: mp_builtin_input at ../py/modbuiltins.c line 245
0x40101b9a: fun_builtin_var_call at ../py/objfun.c line 127
0x400fdf49: mp_call_function_n_kw at ../py/runtime.c line 624
0x40109ef9: mp_execute_bytecode at ../py/vm.c line 919
0x40101c34: fun_bc_call at ../py/objfun.c line 287
0x400fdf49: mp_call_function_n_kw at ../py/runtime.c line 624
0x400e1625: parse_compile_execute at ../lib/utils/pyexec.c line 103
0x400e18c9: pyexec_file at ../lib/utils/pyexec.c line 555
0x400e014c: TASK_Micropython at mptask.c line 320
```

charliewprice, 2020-05-07:
I'm having the issue also - gateway runs for a few hours then hangs requiring a reset.  I'm including a TPL5010 as a watchdog hardware timer.  Software watchdogs simply don't work most of the time.  I wired up the TPL5010 this morning - it's tiny.  It has rebooted the gateway at least 3times today.  So no matter how much improvement can be made in stability -- I don't want to make a trip to reset the gateway when it crashes.  My application is agriculture.

Get a hardware timer today!  :o)

LopPy4 1.20.2rc6