pycom/pycom-libraries #129: 'Loramesh' object has no attribute 'mesh'

plosi, 2020-07-30:
Hello, I am trying to run the pymesh examples but I am getting the following error:
```
Scripts version  4
Traceback (most recent call last):
  File "<stdin>", line 74, in <module>
  File "_pymesh.py", line 57, in __init__
  File "_mesh_interface.py", line 56, in __init__
  File "_mesh_internal.py", line 123, in __init__
  File "_loramesh.py", line 73, in __init__
  File "_loramesh.py", line 122, in _lora_init
AttributeError: 'Loramesh' object has no attribute 'mesh'
```

here is my configuration:

```
Pycom MicroPython 1.20.2.rc11 [v1.11-f4d850b] on 2020-07-08; LoPy4 with ESP32
Pybytes Version: 1.5.2
Pymesh Version: 1.0.1
```

can you help me finding what is wrong? thank you!

catalinio, 2020-07-31:
Hi @plosi,

I'm trying to reproduce your problem. It seem you have the right firmware version.

I've just updated a Lopy4 to the latest v1.20.2.rc10, then add it in Pybytes as new Pymesh project, deployed it and obtained the firmware v.1.20.2.rc11 with Pymesh 1.0.1, same as your configuration.

Next, without any main.py, from REPL I've used:

```
>>> pymesh = pybytes.__pymesh.__pymesh
>>> os.uname()
(sysname='LoPy4', nodename='LoPy4', release='1.20.2.rc11', version='v1.11-f4d850b on 2020-07-08', machine='LoPy4 with ESP32', lorawan='1.0.2', sigfox='1.0.1', pybytes='1.5.2', pymesh='1.0.1')
>>> pymesh.status_str()
"Role 4, Single True, IPv6: ['2001:cafe:cafe:cafe:9d9e:64ac:e351:c7d4', 'fdde:ad00:beef:0:0:ff:fe00:fc00', 'fdde:ad00:beef:0:0:ff:fe00:9c00', 'fdde:ad00:beef:0:0:0:0:7af9', 'fdde:ad00:beef:0:2d6b:182e:7224:71a5', 'fe80:0:0:0:72b3:d549:9a74:7af9']"
```

As I've explained here: https://docs.pycom.io/pymesh/simple-example/, if Pybytes is enabled, then the Pymesh is already initialized. Maybe you try to re-initialise.

plosi, 2020-07-31:
Hi @catalinio,

thank you very much for the prompt reply. I followed your instructions and re-initialiase the node and now it is working perfectly.

Cheers

closed 2020-07-31 by plosi