pycom/pycom-libraries #136: Pytrack 1.0 I2C initialization errors with LoPy4

danalvarez, 2021-03-09:
Hello,

I have been attempting to use the Pytrack 1.0 with DFU 0.0.8 and a LoPy4 running the following firmware:

`(sysname='LoPy4', nodename='LoPy4', release='1.20.2.r0', version='v1.11-783192e on 2020-08-06', machine='LoPy4 with ESP32', lorawan='1.0.2', sigfox='1.0.1', pybytes='1.5.1')`

Unfortunately, initialization of the Pytrack module is spotty at best. I have used both the old pytrack.py library, as well as the pycoproc_1.py library. When I call either `py=Pytrack()` or `py=Pycoproc(Pycoproc.PYTRACK)`, I keep getting I2C errors like these:

```
Traceback (most recent call last):
  File "main.py", line 34, in <module>
  File "/flash/lib/pycoproc_1.py", line 115, in __init__
  File "/flash/lib/pycoproc_1.py", line 115, in __init__
Exception: Board not detected: I2C bus error
```
```
Traceback (most recent call last):
  File "main.py", line 34, in <module>
  File "/flash/lib/pycoproc_1.py", line 118, in __init__
  File "/flash/lib/pycoproc_1.py", line 173, in poke_memory
  File "/flash/lib/pycoproc_1.py", line 134, in _write
OSError: I2C bus error
```

```
Tracback (most recent call last):
  File "main.py", line 6, in <module>
  File "/flash/lib/pytrack.py", line 68, in __init__
  File "/flash/lib/pytrack.py", line 110, in poke_memory
  File "/flash/lib/pytrack.py", line 75, in _write
OSError: I2C bus error
```

If I add a while loop and try/except statements, the Pytrack sometimes initializes after several attempts, and then I am able to use the GPS module to get (lat,lon) coordinates. Basically, once it initializes it does not fail again, but the initialization is extremely difficult to achieve.

For example, when running this code:

```
# See https://docs.pycom.io for more information regarding library specifics

import time

from pytrack import Pytrack
from L76GNSS import L76GNSS

pydone = False
while not pydone:
    try:
        print("trying to initialize Pytrack...")
        py = Pytrack()
        pydone = True
        print("Pytrack init success!")
    except Exception as e:
        print(e)
        time.sleep(2)
l76 = L76GNSS(py, timeout = 60) # GSP timeout set to 60 seconds

while True:
    print(l76.coordinates())
```

I get this output:

```
trying to initialize Pytrack...
I2C bus error
trying to initialize Pytrack...
I2C bus error
trying to initialize Pytrack...
I2C bus error
trying to initialize Pytrack...
Pytrack init success!
(None, None)
(None, None)
(None, None)
```

Which shows that eventually the Pytrack is initialized and the GPS starts to work (eventually I get the correct coordinates instead of (None, None)).

I attempted to read the I2C bus and I got two results. For an immediate initialization of the Pytrack, the SDA line looks like shown below. Note the very beginning seems to be the data sent to initialize the Pytrack and then we receive periodic data from the GPS.

[image]

Then, a delayed initialization (in which the Pytrack only initializes after several attempts) looks like shown below. Note significant more data is sent on the SDA line before the line goes quiet and then the Pytrack initializes (which is shown by the periodic data pulses on the right of the image).

[image]

Another thing that also happens is that sometimes the LoPy4 just hangs, without throwing any exceptions, and then only a power cycle works. This is mentioned in this thread already: https://forum.pycom.io/topic/5595/pytrack-i2c-causes-system-to-freeze

**Any ideas why the initialization of the Pytrack takes so many attempts?** I have already tried to change the LoPy4 to no effect. I cannot change the Pytrack as it is the only module at my disposal.

Best,
Dan

peter-pycom, 2021-03-09:
Hi Dan. 

Wrt the hang - do you actually need to power cycle or is it enough to press the reset button on the lopy? 

Could you do me a favor and put some print statements into the __init__ in pycoproc_1.py to narrow down which line exactly it hangs on? Is it the I2C() at the start or some of the i2c writes later on ...

peter-pycom, 2021-03-09:
Also, one more thing. Could you take photos from your oscilloscope showing the SCL line comparing cases when it works/ doesn't work?

danalvarez, 2021-03-11:
Dear Peter:

Thank you for your very prompt reply.

To answer your question, it's enough to press the reset button, no need for
a complete power cycle.

Unfortunately, due to the urgency of deploying this prototype into the
field, I have moved from using the Pytrack to just using a LoPy4 with an
external GPS module via UART. Much easier than I2C. I will look into the
questions you asked at a later date when I am able to get into that.

But I do appreciate your interest and support.

Best,

Dan

peter-pycom, 2021-03-11:
Ok. I remotely remember that I had seen something like this, but I couldn't reproduce it. So it would be really helpful to have your input. Whenever you have time!