pycom/pycom-libraries #147: MQTT Stops publishing and subscribing after 5 minutes

TanCans, 2021-05-10:
Hello everyone,

i am using the mqtt library to publish some telemetry data to an IoT platform with a pycom microcontroller. But mqtt stops publishing data exact after five minutes even though it is still connected to wlan.

I am sharing the piece of code that calls mqtt client and the library

Connecting wlan:

wlan.connect('my wlan', auth=(None,'my password'), timeout= 50000) #even though i changed this timeout it didnt solve my problem

My credentials (for iot platform)

server = 'my server'
port = 1883
id = 'my id'
pw = ‘acces token’
topic_publish = 'v1/devices/me/telemetry'
dict={}

Creating a client:

Client = MQTTClient(id, server, port, pw, pw)
time.sleep(2)
client.connect()

Publishing in the loop:

While 1
encoded=ujson.dumps(dict)
client.publish(topic_publish, encoded)
time.sleep(1)

Thank you all in advance!

nznobody, 2021-11-25:
Not sure if at all related, but I have occasionally noticed potential memory leaks in the MQTT library or use of it resulting in OSError with out-of-memory.

Whilst not pretty, periodically (e.g. on a thread or async) running `gc.collect()` did the trick...

nznobody, 2022-02-01:
I've since moved to a new (asyncio) implementation since this MQTT library leaves a lot to be desired. It can be found here: https://github.com/nznobody/micropython-umqtt.uasyncio2