pycom/pycom-libraries #150: Faulty code doesn't work

dnear1, 2021-06-17:
https://github.com/pycom/pycom-libraries/blob/b92cea79b32fc89bff4153f5f5f0840a988a52d2/shields/pysense_2.py#L51

says 
print("Light (channel Blue, channel Red): " + str(lt.light()," Lux: ", str(lt.lux()), "lx"))

Should say
print("Light (channel Blue, channel Red): " + str(lt.light() + " Lux: " +  str(lt.lux()) +  "lx"))

Does anyone even test this stuff?

gijsio, 2021-06-17:
I personally tested all the codes and dont know how this mistake still slipped in (you're refering to the missing `)` for `str` right?). I'll correct it, thanks for higlighting!

dnear1, 2021-06-17:
The missing end bracket was at the end of the statement, not after lt.light(), and I had to change the 3 ',' to '+' to get it to run

gijsio, 2021-06-18:
The issue is fixed now. Thanks again. I will close the issue here. Just to confirm, it can work with `+` inside the `str(...)` block, or (`,` and `+`) inside the `print(...)` as a string delimiter. The comma will add spaces, while the plus will directly concatenate strings

closed 2021-06-18 by gijsio